syssi/esphome-solax-x1-mini #11: Similarities to SolarEdge SunSpec protocol ?

mr-sneezy, 2021-09-22:
Anyone know is there any similarity with the Solax protocol over modbus to SunSpec used by other solar inverters including SolarEdge ?
Reason for asking is the possibility to fork this into a ESPhome-modbus-SolarEdge project.

syssi, 2021-09-22:
Do you have any protocol details about the solar edge device?

mr-sneezy, 2021-09-22:
I'm learning about this stuff mostly through the solaredges technical notes like this one.
[https://www.solaredge.com/sites/default/files/sunspec-implementation-technical-note.pdf](url)

syssi, 2021-09-22:
It looks like the SolarEdge has a proper MODBUS implementation. This is the component you are looking for: https://github.com/esphome/esphome/pull/1779