syssi/esphome-solax-x1-mini #15: Solax X1 Mini G2 / Invalid header and Invalid response size

zcloud-at, 2022-02-23:
[15:42:31][VV][modbus_solax:218]: TX -> AA.55.01.00.00.0A.11.02.00.01.1D (11)
[15:42:31][VV][uart.arduino_esp8266:180]:     Flushing...
[15:42:31][VV][modbus_solax:083]: RX <- 00.21.00.02.07.EF.00.00.00.1D.00 (11)
[15:42:31][W][modbus_solax:086]: **Invalid header**.
[15:42:32][VV][scheduler:185]: Running interval 'update' with interval=1000 last_execution=16039 (now=17046)
[15:42:32][VV][modbus_solax:218]: TX -> AA.55.01.00.00.0A.11.02.00.01.1D (11)
[15:42:32][VV][uart.arduino_esp8266:180]:     Flushing...
[15:42:32][VV][modbus_solax:083]: RX <- AA.55.00.0A.01.00.11.82.32.00.21.00.02.07.EC.00.00.00.1D.00.00.00.18.09.55.13.80.02.2B.FF.FF.00.00.5D.AF.00.00.10.50.00.02.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.07.A4 (61)

[15:42:32][W][solax_x1:076]: **Invalid response size: 50**
[15:42:33][VV][scheduler:185]: Running interval 'update' with interval=1000 last_execution=17039 (now=18044)
[15:42:33][VV][modbus_solax:218]: TX -> AA.55.01.00.00.0A.11.02.00.01.1D (11)
[15:42:33][VV][uart.arduino_esp8266:180]:     Flushing...
[15:42:33][VV][modbus_solax:083]: RX <- AA.55.00.0A.01.00.11.82.32.00.21.00.02.07.DB.00.00.00.1D.00.00.00.18.09.56.13.81.02.2C.FF.FF.00.00.5D.AF.00.00.10.50.00.02.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.07.96 (61)

syssi, 2022-02-23:
This

```
RX <- 00.21.00.02.07.EF.00.00.00.1D.00 (11)
```

looks like an incomplete frame. If you compare the payload with the long (61 byte) frames starting with the header `0xAA 0x55` you can fine similar values:

```
AA.55.00.0A.01.00.11.82.32.00.21.00.02.07.EC.00.00.00.1D.00.00.00.18.09.55.13.80.02.2B.FF.FF.00.00.5D.AF.00.00.10.50.00.02.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.07.A4 (61)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

syssi, 2022-02-23:
It looks like the payload has changed a bit. I will try to compare the old status report payload with your payload.

```
# Solax X1 mini
AA.55.00.0A.01.00.11.82..34.00.1A.00.02.00.00.00..00.00.00.00.00.00.00.09..21.13.87.00.00.FF.FF.00..00.00.12.00.00.00.15.00..00.00.00.00.00.00.00.00..00.00.00.00.00.00.00.00..00.00.00.00.00.04.D6
AA.55.00.0A.01.00.11.82.32.00.21.00.02.07.EC.00.00.00.1D.00.00.00.18.09.55.13.80.02.2B.FF.FF.00.00.5D.AF.00.00.10.50.00.02.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.07.A4
# Solax X1 mini g2
```

syssi, 2022-02-23:
```
0xAA 0x55: Header
0x00 0x0A: Source address
0x01 0x00: Destination address
0x11: Control code
0x82: Function code
0x32: Data length (50 bytes)
0x00 0x21: Temperature
0x00 0x02: Energy today
0x07 0xEC: PV1 voltage
0x00 0x00: PV2 voltage
0x00 0x1D: PV1 current
0x00 0x00: PV2 current
0x00 0x18: AC current
0x09 0x55: AC voltage
0x13 0x80: AC frequency
0x02 0x2B: AC power
0xFF 0xFF: unused
0x00 0x00 0x5D 0xAF: Energy total
0x00 0x00 0x10 0x50: Runtime total
0x00 0x02: Mode (2: Normal)
0x00 0x00: Grid voltage fault
0x00 0x00: Grid freq. fault
0x00 0x00: DC injection fault
0x00 0x00: Temperature fault
0x00 0x00: PV1 voltage fault
0x00 0x00: PV2 voltage fault
0x00 0x00: GFC fault
0x00 0x00 0x00 0x00: Error message
0x07 0xA4: CRC
```

--> The last 2 bytes are missing compared to the Solax X1 mini

syssi, 2022-02-23:
Please update your configuration yaml to

```
external_components:
  - source: github://syssi/esphome-modbus-solax-x1@add-solax-x1-mini-g2-compatibility
    refresh: 0s
```

and recompile + flash your ESP. The payload of your device should be accepted/parsed now.

zcloud-at, 2022-02-23:
Thank you for your quick feedback. 
I tested the new version and the error message doesn't seem to appear anymore. Unfortunately the sun is gone now and the X1 MINI won't start anymore. I'll give you a reply tomorrow..

`[17:48:13][VV][scheduler:185]: Running interval 'update' with interval=1000 last_execution=34854 (now=35854)
[17:48:13][VV][modbus_solax:218]: TX -> AA.55.01.00.00.0A.11.02.00.01.1D (11)
[17:48:13][VV][uart.arduino_esp8266:180]:     Flushing...
[17:48:13][VV][modbus_solax:083]: RX <- AA.55.00.0A.01.00.11.82.32.00.17.00.06.07.C6.00.00.00.00.00.00.00.00.09.1E.13.88.00.00.FF.FF.00.00.5D.B3.00.00.10.52.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.06.EB (61)
[17:48:13][V][text_sensor:016]: 'solar-powermeter errors': Received new state 
[17:48:13][D][text_sensor:067]: 'solar-powermeter errors': Sending state ''
[17:48:13][V][text_sensor:016]: 'solar-powermeter mode name': Received new state Wait
[17:48:13][D][text_sensor:067]: 'solar-powermeter mode name': Sending state 'Wait'
[17:48:14][VV][scheduler:185]: Running interval 'update' with interval=1000 last_execution=35854 (now=36854)
[17:48:14][VV][modbus_solax:218]: TX -> AA.55.01.00.00.0A.11.02.00.01.1D (11)
[17:48:14][VV][uart.arduino_esp8266:180]:     Flushing...
[17:48:14][VV][modbus_solax:083]: RX <- AA.55.00.0A.01.00.11.82.32.00.17.00.06.07.C6.00.00.00.00.00.00.00.00.09.1D.13.85.00.00.FF.FF.00.00.5D.B3.00.00.10.52.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.06.E7 (61)`

syssi, 2022-02-23:
You have to use three back ticks instead of one for nice code blocks. ;-) 

The no errors state and operation mode "Wait" looks good already! Could you provide some details (type label) about your inverter? I would like to mention the supported device at the documentation.

zcloud-at, 2022-02-24:
Communication is now stable throughout the day.

```ruby
[17:03:11][D][text_sensor:067]: 'solar-powermeter errors': Sending state ''
[17:03:11][V][text_sensor:016]: 'solar-powermeter mode name': Received new state Normal
[17:03:11][D][text_sensor:067]: 'solar-powermeter mode name': Sending state 'Normal'
[17:03:12][VV][scheduler:185]: Running interval 'update' with interval=1000 last_execution=10953068 (now=10954072)
[17:03:12][VV][modbus_solax:218]: TX -> AA.55.01.00.00.0A.11.02.00.01.1D (11)
[17:03:12][VV][uart.arduino_esp8266:180]:     Flushing...
[17:03:12][VV][modbus_solax:083]: RX <- AA.55.00.0A.01.00.11.82.32.00.1A.00.3D.07.08.00.00.00.03.00.00.00.06.09.0F.13.8A.00.3B.FF.FF.00.00.5D.F0.00.00.10.5B.00.02.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.06.E6 (61)
```

I use a SolaX X1 mini inverter model X1-2.0-S-D(L).
- master version 1.08
- manager version 1.07

syssi, 2022-02-24:
Perfect! You should decrease the log level now. ;-) Printing UART traffic is expensive. I will merge the feature branch to `main` now. Please update your `external_component` section to:

```
external_components:
  - source: github://syssi/esphome-modbus-solax-x1@main
    refresh: 0s
```

PR #16 (2022-02-24): Add Solax X1 Mini G2 support
Closes: #15

```
external_components:
  - source: github://syssi/esphome-modbus-solax-x1@add-solax-x1-mini-g2-compatibility
    refresh: 0s
```

SolaX X1 mini inverter model X1-2.0-S-D(L). Master version 1.08, Manager version 1.07.
closed 2022-02-24 by syssi