syssi/esphome-solax-x1-mini #18: [solax_x1:188]: The device is or was offline. Broadcasting discovery for address configuration...

beocycris, 2022-03-13:
Hey guys,

thank for this modules. In my case I try to use it with a Solax X1 mini 1.5 V2.
And I only receive the text which is mentioned in the title.
May anyone has the same issue and can help me to fix it.

regards
frank

syssi, 2022-03-13:
Could you provide some details about your setup? A photo of your wiring (incl. the RS485 module) and the important parts (`uart` secontion) of your configuration yaml?

Another intepretation of the log message is: "The device doesn't respond". Therefore we must sure your device is connected properly. If this is the case it's possible your device version uses another protocol. Let's start with the wiring!

beocycris, 2022-03-13:
I won´t be at home till friday - so I decided to do a small scheme of the cabling.

[solax_issue.pdf](https://github.com/syssi/esphome-modbus-solax-x1/files/8240550/solax_issue.pdf)

And the code too:

`
substitutions:
  name: solar-powermeter

esphome:
  name: ${name}
  platform: esp8266
  board: d1_mini

external_components:
  #- source: github://syssi/esphome-modbus-solax-x1@add-solax-x1-mini-g2-compatibility
  - source: github://syssi/esphome-modbus-solax-x1@main
    refresh: 0s

uart:
  baud_rate: 9600
  tx_pin: GPIO2
  rx_pin: GPIO1

solax_x1:
  serial_number: "3132333435363737363534333231"
  address: 0x0A
  update_interval: 2s

text_sensor:
  - platform: solax_x1
    mode_name:
      name: "${name} mode name"
    errors:
      name: "${name} errors"
`

thanks!

syssi, 2022-03-14:
I assume your RS485 module isn't handled properly:

https://electronics.stackexchange.com/questions/153500/correctly-using-re-and-de-with-rs485

I will add a flow control pin to the component to support your rs485 module. At the moment DE/RE isn't used correctly.

syssi, 2022-03-14:
I've introduce a flow control pin here:

https://github.com/syssi/esphome-modbus-solax-x1/pull/19

Please use this new feature branch for your tests:

```
external_components:
  - source: github://syssi/esphome-modbus-solax-x1@add-flow-control
    refresh: 0s
```

The next step is to update your configuration to use flow control pin:

https://github.com/syssi/esphome-modbus-solax-x1/pull/19/files#diff-dbc81de7d9fb76af6e67b9c8b02f69c8ea7f35b243a2898f26520c759e0d7b6d

This should fit for you

```
uart:
  id: uart0
  baud_rate: 9600
  tx_pin: GPIO2
  rx_pin: GPIO1

modbus_solax:
  id: modbus0
  uart_id: uart0
  flow_control_pin: GPIO2

solax_x1:
  modbus_solax_id: modbus0
  serial_number: "3132333435363737363534333231"
  address: 0x0A
  update_interval: 2s
```

beocycris, 2022-03-14:
Do i have to change my wiring physically? Or may I can use the:
D1=GPIO4 as `tx` to the RS485-pin `RO`,
D2=GPIO5 as `rx` to the RS485-pin `DI` and 
GPIO0 as flowcontrole pin, connected to the bridged RS485-pins `RE/DE` ?

Or shell I change GPIO5 to the original `tx`-pin and the GPIO to the `rx`-pin on the microcrontroller?

Because I changed the my code to your provided one and nothing different happened than the issue above.

syssi, 2022-03-14:
The `tx_pin` must connected to `DI` **not** `RI`. If you use `GPIO4` and `GPIO5` a software serial implementation is used by ESPHome. I would use the hardware UART pins of the ESP8266. The best wiring would look like this:

https://github.com/KlausLi/Esp-Soyosource-Controller/blob/main/BastelPlan3000_Soyosource_Controller_by_BavarianSuperGuy.png

```
api:
ota:
logger:
  level: INFO
  baud_rate: 0  # <--- super important!

uart:
  id: uart0
  baud_rate: 9600
  tx_pin: GPIO1
  rx_pin: GPIO3

modbus_solax:
  id: modbus0
  uart_id: uart0
  flow_control_pin: GPIO0

solax_x1:
  modbus_solax_id: modbus0
  serial_number: "3132333435363737363534333231"
  address: 0x0A
  update_interval: 2s
```

beocycris, 2022-03-14:
Ah ok got you.

So my config: 

uart:
   id: uart0
   baud_rate: 9600
   tx_pin: GPIO4
   rx_pin: GPIO5

modbus_solax:
   - id: modbus0
     uart_id: uart0
     flow_control_pin: GPIO2

solax_x1:
   modbus_solax_id: modbus0
   serial_number: "3132333435363737363534333231"
   address: 0x0A
   update_interval: 2s

Won´t work properly.
Then I really have to wait till weekend, when I´m able to solder the pins to `tx` and `rx` instead using `GPIO4` and `GPIO5`. `GPIO2` also seems to be wrong and has to be `GPIO0`.

beocycris, 2022-03-14:
1.: Wow - tell me "the stupid guy".
2.: Yes I´m from Germany.
3.: You're right.
4.: See 1. ......

So I have to give it another try tomorrow!

beocycris, 2022-03-15:
I´m just trying to receive the data stream. But like yesterday in the evening, no changes .....

syssi, 2022-03-15:
Please check your wiring this weekend. It's super important `RE/DE` is connected properly.

Inverter settings: I don't have any menu items to enable/disable modbus. There is also no menu item to change the modbus address / device id for example. At my firmware version modbus is always enabled. Are you able to control something in this regard?

beocycris, 2022-03-15:
Yes - I will check, if `RE/DE` is connected properly

I have the enabling option and also a menu to change the address of modbus. 

But like you said, I´m only able to check that on friday.

syssi, 2022-03-15:
The different menu sounds like real modbus rtu (polling selected registers). There are multiple things we should try as soon you have access to your device. ;-)

syssi, 2022-03-16:
Could you provide your exact model name of the type label (something like `X1-2.0-S-D(L)`), the software versions (master/manager) of your device and may be some photos of the menu structure? It should help to compare your device with well known devices/protocols.

beocycris, 2022-03-19:
Good morning. Well- the sun is shining so I was able to take a look at my settings. 

[image]
[image]
[image]

syssi, 2022-03-19:
Could you look for some firmware versions at the menu?

beocycris, 2022-03-19:
[image]
[image]

Do you need these?

syssi, 2022-03-19:
Perfect! Next wish: Could you provide a photo of the type label (on the right side of the device)?

syssi, 2022-03-19:
Please provide some photos of your new wiring as next step. Enable the debug output of the UART component:

https://github.com/syssi/esphome-modbus-solax-x1#debugging

Provide some logs. It would like to know there is some serial traffic (outgoing & incoming) or not.

beocycris, 2022-03-20:
Sorry for the bad quality:
`TX`= `DI` = blue cable
`RX` = `RO` = green cable
 `D4` = `DE/RE` = brown cable
[image]
[image]

syssi, 2022-03-20:
Alright! Next step is to collect some serial traffic. Could you provide some logs (debug output of the `uart` component enabled)?

syssi, 2022-03-20:
Could you provide your configuration yaml? I would like to check you are using (D4 = GPIO2 vs. D3 = GPIO0).

beocycris, 2022-03-20:
[image]

I think there ist some communication......

beocycris, 2022-03-20:
[image]

syssi, 2022-03-20:
Looks good! :-) Issue solved.

beocycris, 2022-03-20:
Thank you very much for supporting me!

syssi, 2022-03-20:
Please make sure to set the `logger` to 0 baud if you use GPIO1 and 3. Otherwise log messages are transmitted to the inverter too. ;-)

syssi, 2022-03-20:
I will merge the flow control support into main now. Please update the feature branch at the external components section to `@main` again later the day.

beocycris, 2022-03-20:
[image]

Of course. It should be fine like this!

syssi, 2022-03-20:
Perfect! Last question about your device: You've enabled the ModbusRTU setting and the `device address` is set to `1`. Right? I would like to improve the docs.

closed 2022-03-20 by syssi

beocycris, 2022-03-20:
No is disabled the RTU option. 
[image]
[image]

commit a900cf27 (2022-03-20): Introduce flow control pin (Closes: #18) (#19)