syssi/esphome-solax-x1-mini #20: just question about Solax Smart Meter

straga, 2022-03-22:
Maybe you have communication protocol Invertor <--> Meter

I am investigating:

Solax X1 mini:
That ask every second like a ping. That works.
send solax: b'\x01\x03\x00\x0e\x00\x01\xe5\xc9'
meter SDM230 not from Solax anw: b'\x01\x03\x02\x00\x00'

send solax: b'\x01\x03\x00\x0b\x00\x01\xf5\xc8'
meter SDM230 not from Solax anw: b'\x01\x03\x02\x00\x00'

But that probably Solax request Data:
send solax: b'\x01\x03\x00\x08\x00\x04\xc5\xcb'
meter SDM230 not from Solax anw: b'\x01\x03\x08\x00\x00\x00\x00\x00\x00\x00\x00'

Meter answer: 00 00 00 00 00 00 00 00
I do STM32 rs485 gateway, but need to know from Original Solax Meter what answer.

Maybe someone has that information? Thanks.

syssi, 2022-03-22:
Could you describe the goal of your STM RS485 gateway?

straga, 2022-03-22:
Cople days ago I bought a Solar panel. Last year I bought Solax X1 mini and SDM230modbus not Solax. 
Today I am all connected together that work. Invertor can seed meter but always get 0 watts.
I have standart SDM230 meter. I try to buy Solax Smart Meter - it is possible after a month or more.

I do that. One side connects to Solax. Second to Meter. I get Modbus messages.
My idea is just to convert the message to Solax protocol. But I don't have information about that.

1. step - 00 0e - in manual not a description about that register.

Solax -> meter: 01 03 00 0e 00 01 - 1: devid, 03: read from Read Holding Registers, 00 0e = 14, +1 = 40015, 1: count register
Meter -> Solax: 01 03 02 00 00 - 2 byte with 00 00 as value.
That ok Solax accept that.

2. step - 00 0b - in manual not a description about that register. 
 Solax -> meter: 01 03 00 0b 00 01 - 1: devid, 03: read from Read Holding Registers, 00 0b = 11, +1 = 40012, 1: count register
Meter -> Solax: 01 03 02 00 00 - 2 byte with 00 00 as value.
That ok Solax accept that.

3. step - that ask every second 3 times.  
Solax -> meter: 01 03 00 08 04 - 1: devid, 03: read from Read Holding Registers,  00 08 = 8, 4009-40010-4011-4012, 4: count register
Meter -> Solax: 01 03 08 00 00 00 00 00 00 00 00, 8 byte, 4009 = 00 00, 4010 = 00 00, 4011 = 00 00, 4012 = 00 00.

Idea get from Solax request, Ask from Meter information = And Reply to Solax Correct Data like a Solax Smart Meter.

My RS485 parser.
[image]

syssi, 2022-03-22:
Got it. You are trying to implement a protocol converter to be able to talk to an incompatible SDM230 firmware.

Last thing I would like to understand: What's the purpose of the SDM230? Do you export power to the grid and the SDM230 should count the amount for billing or do you use the SDM230 as display only? There is no way to control the Solax Mini via the SDM230 (to avoid exporting to the grid), right?

straga, 2022-03-22:
I want to set Export Zero == 0. I can't export because City Grid Meter counts my export like IMPORT. I need always Zero export.
X1 Mini - need Meter for that. X1 mini ask meter every second and set how much need watt generate for Zero export. 

Yess. Need graap message Solax Smart Meter or get information about that from some documentations.

syssi, 2022-03-22:
Could you tell me your exact Solax inverter model?

And you are right: The payload (SDM230 -> Solax Mini) to control the Solax Mini isn't known. We need some captures of a SDM230 (using the special solax firmware version) owner here. I could provide some instructions how to capture the traffic using a ESPHome node.

straga, 2022-03-22:
My model: X1-2.0-S-D(L), Gen2 with latest firmware from support.
[solax_X1_firmware_update.zip](https://github.com/syssi/esphome-modbus-solax-x1/files/8323182/solax_X1_firmware_update.zip)

[image]

Yes, will be a cool ESPhome RS485 traffic capture.

I do on micropython traffic capture.
For capture: 
I use simple UART code, just read all traffic on RS485.

For emulation:
Use: https://github.com/brainelectronics/micropython-modbus

User STM32 Weact with micropython that board has 3xUART.
One UART over RS485 as Slave connect to Solax X1 mini.
Second UART over RS485 as Mater to SDM230.

syssi, 2022-03-22:
You can do the same using this esphome configuration:

```
substitutions:
  name: uart-test-device1
  tx_pin: GPIO1
  rx_pin: GPIO3

esphome:
  name: ${name}
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

api:
ota:
logger:
  baud_rate: 0
  level: DEBUG

uart:
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
    dummy_receiver: true
    after:
      timeout: 100ms
    sequence:
      - lambda: UARTDebug::log_hex(direction, bytes);

#interval:
#  - interval: 2s
#    then:
#      - uart.write: "${name}\r\n"
```

syssi, 2022-03-22:
Some details about the SDM230: https://www.jwsmartmeters.co.uk/wp-content/uploads/2018/02/Eastron-SDM230-Mbus-protocol-V1.0.pdf

straga, 2022-03-22:
That Mbus. But we use Modbus.
[eastron-sdm230-protocol-v14.pdf](https://github.com/syssi/esphome-modbus-solax-x1/files/8323539/eastron-sdm230-protocol-v14.pdf)

syssi, 2022-03-22:
@straga Is this interpretation of your first message correct:

```
TX>>> 01:03:00:0e:00:01:e5:c9 (Request 1 from Solax)
RX<<< 01:03:02:00:00 (Response 1 from non-solax SDM230)

TX>>> 01:03:00:0b:00:01:f5:c8 (Request 2 from Solax)
RX<<< 01:03:02:00:00 (Response 2 from non-solax SDM230)
      
TX>>> 01:03:00:08:00:04:c5:cb (Request 3 from Solax)
RX<<< 01:03:08:00:00:00:00:00:00:00:00 (Response 3 from non-solax SDM230)
```

Do you see all 3 requests + responses every second?

syssi, 2022-03-22:
Just a guess: 

https://github.com/arendst/Tasmota/blob/development/tasmota/xnrg_21_sdm230.ino#L221-L223
https://github.com/arendst/Tasmota/blob/development/tasmota/xnrg_21_sdm230.ino#L253-L255

1. Request: Phase angle
2. Request: Power demand
3. Request: Total energy

syssi, 2022-03-22:
You could try to emulate the responses of the SDM. I assume the most important response is the power demand. For request 1 and 3 you could respond with static payload. A bit strange: The responses doesn't have a checksum.

straga, 2022-03-22:
I am removing CRC in my example. Just for a clear picture. Ofcource CRC exists. 
I will try tomorrow. When sunrise.

straga, 2022-03-22:
CRC I calculate in the last step before sending to uart. 
https://github.com/brainelectronics/micropython-modbus/blob/948c1a23f8ff5550a7e323b27bff04f926f747cd/umodbus/serial.py#L56

If the will is not correct in CRC - The invertor immediately stops work and shows an error Meter Fault.

Tomorrow I give you the full TX<->RX

straga, 2022-03-23:
That data from Solax Inverter to Meter. Solax things All right and get 0 watts.
Log when Activate Meter in SOLAX invertor. 

```
 << uart(solax: 01 03 00 0e 00 01 e5 c9 - frame
 << uart(solax): 01 03 00 0e 00 01 e5 c9 - data
 >> uart(solax) : 01 03 02 00 00 b8 44

 << uart(solax: 01 03 00 0e 00 01 e5 c9 - frame
 << uart(solax): 01 03 00 0e 00 01 e5 c9 - data
 >> uart(solax) : 01 03 02 00 00 b8 44

 << uart(solax: 01 03 00 0e 00 01 e5 c9 - frame
 << uart(solax): 01 03 00 0e 00 01 e5 c9 - data
 >> uart(solax) : 01 03 02 00 00 b8 44

 << uart(solax: 01 03 00 0e 00 01 e5 c9 - frame
 << uart(solax): 01 03 00 0e 00 01 e5 c9 - data
 >> uart(solax) : 01 03 02 00 00 b8 44

 << uart(solax: 01 03 00 0e 00 01 e5 c9 - frame
 << uart(solax): 01 03 00 0e 00 01 e5 c9 - data
 >> uart(solax) : 01 03 02 00 00 b8 44

 << uart(solax: 01 03 00 08 00 04 c5 cb - frame
 << uart(solax): 01 03 00 08 00 04 c5 cb - data
 >> uart(solax) : 01 03 08 00 00 00 00 00 00 00 00 95 d7

 << uart(solax: 01 03 00 08 00 04 c5 cb - frame
 << uart(solax): 01 03 00 08 00 04 c5 cb - data
 >> uart(solax) : 01 03 08 00 00 00 00 00 00 00 00 95 d7

 << uart(solax: 01 03 00 08 00 04 c5 cb - frame
 << uart(solax): 01 03 00 08 00 04 c5 cb - data
 >> uart(solax) : 01 03 08 00 00 00 00 00 00 00 00 95 d7

 << uart(solax: 01 03 00 0e 00 01 e5 c9 - frame
 << uart(solax): 01 03 00 0e 00 01 e5 c9 - data
 >> uart(solax) : 01 03 02 00 00 b8 44

```

syssi, 2022-03-23:
Alright. As next step you should try to respond (periodically) to `01 03 00 0e 00 01 e5 c9` with a fixed frame telling the inverter to export 100W:

```
01 03 02 00 64 CRC1 CRC2
```

straga, 2022-03-23:
Done all works.
Get Frame from Solax with 0e, response INT with sign- Watts.
get from meter -220.34 convert to -220. Send to SOLAX.

[image]
if activate Meter but export value limit = 0 watts
```
 << uart(solax: 01 03 00 0e 00 01 e5 c9 - frame
 << uart(solax): 01 03 00 0e 00 01 e5 c9 - data
func:3
val: [0]
 >> uart(solax) : 01 03 02 00 00 b8 44
 >> uart(meter) : 01 04 00 0c 00 02 b1 c8
 << uart(meter) : 00 00 00 00 = 0.0
Status of hreg 1: [0]
```

if activate Meter but export value limit = 20000 watts
```
 << uart(solax: 01 03 00 0e 00 01 e5 c9 - frame
 << uart(solax): 01 03 00 0e 00 01 e5 c9 - data
func:3
val: [418]
 >> uart(solax) : 01 03 02 01 a2 38 6d
 >> uart(meter) : 01 04 00 0c 00 02 b1 c8
 << uart(meter) : c3 d1 17 a9 = -418.1849
Status of hreg 1: [418]
 >> uart(meter) : 01 04 00 0c 00 02 b1 c8
 << uart(meter) : c3 d1 17 a9 = -418.1849
Status of hreg 1: [418]
```

syssi, 2022-03-23:
I don't understand the impact of the export value limit (0 vs 20000 watts) yet.

straga, 2022-03-23:
Export value = 20000 that more than Inverter can generate watt and that means meter connect and not work like a limiter, just shows Data. 
If set 0 or 100watt that is lower than inverter can generate and can work like Limiter.

syssi, 2022-03-23:
Could you share your code somewhere? I assume your register address mapping would be helpful for people facing the same problem.

straga, 2022-03-23:
I am putting the source as is on Today: 
 - https://github.com/straga/Smart-Meter-Gateway

syssi, 2022-03-23:
Let's close this issue. :-) It's solved.

closed 2022-03-23 by syssi

straga, 2022-03-23:
We are Done.
Thanks, Sebastian Muszynski for the help.

tuxmike, 2022-04-02:
Hi, do you know, which message set the grid power to 336W in the solax panel screenshot above?

What exactly does the value 418u (0x01a2) sent with 
`01 03 02 01 a2 38 6d`
from register 0x0e represent?

Is it measured power or some power target?

If it's watts, how is the opposite current flow indicated? MSB bit (two's complement)?

Thanks

syssi, 2022-04-02:
> Hi, do you know, which message set the grid power to 336W in the solax panel screenshot above?

```
01 03 02 WATT1 WATT2 CRC1 CRC2
```

> What exactly does the value 418u (0x01a2) sent with `01 03 02 01 a2 38 6d` from register 0x0e represent?

I don't know the correct answer. This is my guess: The smartmeter imports 418 watts from the grid at the moment. If you send this power demand to the solar inverter the inverter will try to provide the 418 watt possible.

> Is it measured power or some power target?

Good question. The bytes at the capture above aren't signed. 

> If it's watts, how is the opposite current flow indicated? MSB bit (two's complement)?

@straga Could you help here / provide some example payloads? In the payload above the case is missing.

straga, 2022-04-02:
I do some update code. Only what need and asyncio. 
https://github.com/straga/Smart-Meter-Gateway/blob/uasyncio_version/modbus_uart/runner.py

that ask from meter.
https://github.com/straga/Smart-Meter-Gateway/blob/32de2747229074207fdbf229bddb14b58473955b/modbus_uart/runner.py#L121

that answer to Solax
https://github.com/straga/Smart-Meter-Gateway/blob/32de2747229074207fdbf229bddb14b58473955b/modbus_uart/runner.py#L150

Solax just ask power in watt (real time demand) what meter detect - power demand from grid

https://user-images.githubusercontent.com/34532/161398652-cdd9ea02-2a0c-4d91-ab53-57b5efe95aaa.mp4

syssi, 2022-04-03:
@straga Could you verify/explain the following scenario:

1. Your freezer starts (requires 800W)
2. The smartmeter requests the demand of 800W at the inverter
3. The inverter push the demand to the grid
4. The new demand is `0` watts no because the solar is pushing 800 watts.
5. The freezer turns off
6. The new demad is -800 watts
7. What does your converter/gateway now? Is the calculated new demand 800 watts, 0 watts or -800 watts?

straga, 2022-04-03:
Gateway just sends watt from the meter and converts value for format Solax understands.
If meter 
- meter shows  800w from GRID City    - 
GTW send 800
- meter shows -800w  from GRID City  -  GTW send -800 
- meter shows   0    from GRID City      - 
GTW send 0  

Gateway every 0.1 second get watts(from GRID City) from the meter and write to register dictionary like float.
Solax asks Gateway when needed Solax, Gateway read from 
register dictionary float -> convert to int and make the response to Solax.

If on the Solax inverter Turn on - Export Control, Solax will be a calc value less or == to export value.
if set export control limit value:0 watt.

example: without limit set all extra generate watts send to the Grid City.
- Limit export 20000 watts
- PVarray can generate 800watts.
- Air-condition load 1000watt.
- Light load 100watts.
= Meter shows = 300 watts from Grid City. get watts to the Grid City

example: without limit set all extra generate watts send to the Grid City.
- Limit export 20000 watts
- PVarray can generate 800watts.
- Air-condition load 400watt.
- Light load 100watts.
= Meter shows = -400 watt from Grid City. send watts to Grid City

example: with limit 0 watts. Not send any watts to the Grid City.
- Limit export 0 watts
- PVarray can generate 800watts.
- Air-condition load 400watt.
- Light load 100watts.
= Meter shows = -400 watt from Grid City. send watts to Grid City
Solax calc self-generator set to 500 watt
- PVarray set generates 500watts.
- Air-condition load 400watt.
- Light load 100watts.
= Meter shows = 0 watts from Grid City. send/get watts to Grid City

tuxmike, 2022-04-03:
@straga ah, ok, thanks for explaining :+1:. So the 0e / 40015 / request payload is a singed short ("h") and represent watts at meter:
- positive value: power from city grid
- negative value: power to city grid

One question still:
 Did you find out, what the 0x0b (request 2) and 0x08 (request3) payloads mean?
- request 2 seems to happen only at startup and returning 0x0000 works for you?
- request 3 seems to happen every 30s and is maybe some cumulated kwh value query, that can be ignored?

straga, 2022-04-03:
1. Yes

2. request 2 seems to happen only at startup and returning 0x0000 - yes that works.
3. request 3 seems to happen every 30s and is maybe some cumulated kwh value query, that can be ignored? Yes, I am ignoring it.

May be in someone have original meter and help us understand 2 and 3. Anyway all work like that now.

syssi, 2022-04-03:
Good job guys! I've a clear picture of the converter/gateway now!

rand12345, 2022-05-04:
> May be in someone have original meter and help us understand 2 and 3. Anyway all work like that now.

Hi, Solax SDM230 owner here. This is connected to a Solax X1 Gen3 and hardwired with CAT5. Everything I have tried has lead to the Solax inverter in an alarm state but running this passthrough code works fine. There needs to be a lot of registers made available to make this work and I'll start adding them tomorrow.

https://github.com/rand12345/Smart-Meter-Gateway/blob/uasyncio_version/modbus_uart/passthrough.py

I've used your latest asyncio code to capture this traffic, I'm really new to modbus so this has helped enormously. Hope it can be of some use in the community.

Note that my inverter firmware can address two SDM230s on the same link [01,02]

```
145ms Solax >> Meter: 01 03 00 0b 00 01 f5 c8
209ms Meter >> Solax: 01 03 02 00 a8 b9 fa
409ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
669ms Solax >> Meter: 01 04 00 48 00 02 f1 dd
729ms Meter >> Solax: 01 04 04 46 2a 75 21 29 8c
921ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
1189ms Solax >> Meter: 01 04 00 48 00 02 f1 dd
1249ms Meter >> Solax: 01 04 04 46 2a 75 21 29 8c
1440ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
1709ms Solax >> Meter: 01 04 00 48 00 02 f1 dd
1769ms Meter >> Solax: 01 04 04 46 2a 75 22 69 8d
1959ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
2219ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
2279ms Meter >> Solax: 01 04 04 45 8e 3c 35 5f b4
2475ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
2739ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
2799ms Meter >> Solax: 01 04 04 45 8d de 69 e7 2d
2994ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
3259ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
3319ms Meter >> Solax: 01 04 04 45 8d e6 b6 b5 75
3512ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
3779ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
3839ms Meter >> Solax: 01 04 04 45 8d e7 42 b5 62
4030ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
4299ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
4359ms Meter >> Solax: 01 04 04 45 8d c8 ef 68 ef
4549ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
4809ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
4869ms Meter >> Solax: 01 04 04 45 8d fb c2 bc 02
5065ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
5329ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
5389ms Meter >> Solax: 01 04 04 45 8d f1 59 fb 09
5584ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
5849ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
5909ms Meter >> Solax: 01 04 04 45 8d d1 cb 63 64
6102ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
6369ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
6429ms Meter >> Solax: 01 04 04 45 8d b2 0e 8b c7
6619ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
6889ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
6949ms Meter >> Solax: 01 04 04 45 8d e8 93 70 ce
7139ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
7399ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
7459ms Meter >> Solax: 01 04 04 45 8d b2 4a 8b f4
7656ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
7919ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
7979ms Meter >> Solax: 01 04 04 45 8d bc c2 8f f2
8174ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
8439ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
8499ms Meter >> Solax: 01 04 04 45 8e 50 cb f3 34
8692ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
8959ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
9019ms Meter >> Solax: 01 04 04 45 8d e7 f3 75 16
9210ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
9482ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
9549ms Meter >> Solax: 01 04 04 45 8d d6 59 e0 f9
9738ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
9988ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
10049ms Meter >> Solax: 01 04 04 45 8d 33 97 2b fd
[... 105 lines cut ...]
28376ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
28639ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
28699ms Meter >> Solax: 01 04 04 45 8a 1a 8e 44 66
28894ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
29159ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
29219ms Meter >> Solax: 01 04 04 45 89 c5 82 ec 53
29413ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
29679ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
29739ms Meter >> Solax: 01 04 04 45 87 02 c5 9f 92
29930ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
30199ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
30259ms Meter >> Solax: 01 04 04 45 86 7d 80 2f 91
30449ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
30709ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
30769ms Meter >> Solax: 01 04 04 45 83 00 2e 9f 7c
30966ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
31229ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
31289ms Meter >> Solax: 01 04 04 45 82 c4 44 1c 53
31484ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
31749ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
31809ms Meter >> Solax: 01 04 04 45 83 56 b5 e1 77
32003ms Solax >> Meter: 02 03 00 0b 00 01 f5 fb
32269ms Solax >> Meter: 01 04 00 0c 00 02 b1 c8
32329ms Meter >> Solax: 01 04 04 45 80 9c 41 46 50
```

straga, 2022-05-05:
@rand12345 
That is very good. We can do Gateway like the original meter.
If you have all original. Can you test:

- if AC Load is under 50w, maybe 20w for example, Sun power MAX. Export Control is set to 0. How many watts inverter is exported?

I have: if under 50w - inverter exported 12-20w always. If AC load to 100w - Export always 0w. Sun power MAX.

rand12345, 2022-05-05:
Unfortunately I can never get my AC load that low to test. Also, my home setup has many inverters and a high load at all times.

I do know that inverters do not operate well at low levels when throttling and what you describe is normal. I have many inverters here that limit export and anything under 100W generated is difficult and unstable. If your goal is to never export energy then I think you will struggle. There will always be times where energy is exported even with export limits. Consider your loads, high frequency switched loads will always be difficult to react to.

CCALLEJAR, 2023-05-12:
Hello, I want to emulate the sdm230 to inject kw into the network of my battery, I have Homeassistant with Esphome with esp32 module and TTL RS485 module. But my programming is limited, I don't know micropython. It is possible to have it in ESPHOME with number: - platform: template => name: "W consumed".
I have also found:
https://github.com/That-Dude/esphome-fake-eastron-SDM230
But I haven't been able to get it to work.

Thank you so much.

syssi, 2023-05-12:
Could you describe your setup and your goal?

Did you see this solution? https://github.com/syssi/esphome-solax-x1-mini/blob/main/esp8266-meter-gateway.yaml

CCALLEJAR, 2023-05-12:
Thank you very much, I will try the solution.
I have a Deye connected to an Eastron sdm230-modbus. The objective is to disconnect them, and connect Esphome (esp32 +TTL RS485)= (virtual Eastron sdm230) to the deye. From Homeassistant decide to download the battery to the network or keep the sdm230 data.
This is how I manage to compensate consumption peaks with the hourly balance or even when the sale price is high and the battery is full and the sun forecast. When there is sun the sale price is very low.

CCALLEJAR, 2023-05-12:
The sdm230 data if I extract them with Esphome, if I disconnect it from the deye. https://esphome.io/components/sensor/sdm_meter.html

syssi, 2023-05-12:
It looks like the initially mentioned project is what you are looking for: https://github.com/That-Dude/esphome-fake-eastron-SDM230

My implementation (`solax_meter_gateway`) is able to respond to power demand requests of a Solax X1 mini only. The official `sdm_meter` component of the ESPHome core is a client implementation to retrieve values from a SDM230.

May be this clarification/detail is helpful for you: A Modbus consists of a single **master** talking to one or multiple **slaves**. The official `sdm_meter` implementation using the `modbus_controller` forms a Modbus Master. I guess in your setup the Deye inverter is the Modbus master and you have to provide a slave / client implementation.

CCALLEJAR, 2023-05-12:
Totally agree with the comments.

Looks like https://github.com/That-Dude/esphome-fake-eastron-SDM230 is in progress => "** This is a work in progress **"

My programming is limited and I don't know how to continue
I have tried to connect it, but without success.