syssi/esphome-solax-x1-mini #22: Solax x1 mini G3

neujbit, 2022-07-27:
i have a problem with the Solax x1 mini G3 with  following errors
[07:18:49][W][solax_x1:083]: Invalid response size: 56
[07:18:49][W][solax_x1:085]: Your device is probably not supported. Please create an issue here: https://github.com/syssi/esphome-modbus-solax-x1/issues
[07:18:49][W][solax_x1:087]: Please provide the following status response data: 00.17.00.00.03.DE.00.00.00.08.00.00.00.05.09.06.13.85.00.48.FF.FF.00.00.00.28.00.00.00.0C.00.02.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.01.00.8A.00.DE (56)

syssi, 2022-07-27:
I will try to decode the data later the day and will provide some feedback your device can be easily supported or not.

syssi, 2022-07-27:
Your device can be supported easily. I will prepare a pull request now:

```
> 00.17.00.00.03.DE.00.00.00.08.00.00.00.05.09.06.13.85.00.48.FF.FF.00.00.00.28.00.00.00.0C.00.02.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.01.00.8A.00.DE (56)

Data  0...1  (temperature):        0x00 0x17 (23 C)
Data  2...3  (energy today):       0x00 0x00 (0.0 kWh)
Data  4...5  (pv1 voltage):        0x03 0xDE (99.0 V)
Data  6...7  (pv2 voltage):        0x00 0x00 (0.0 V)
Data  8...9  (pv1 current):        0x00 0x08 (0.8 A)
Data 10...11 (pv2 current):        0x00 0x00 (0.0 A)
Data 12...13 (ac current):         0x00 0x05 (0.5 A)
Data 14...15 (ac voltage):         0x09 0x06 (231.0V)
Data 16...17 (ac frequency):       0x13 0x85 (49.97 Hz)
Data 18...19 (ac power):           0x00 0x48 (72 W)
Data 20...21 (unused):             0xFF 0xFF
Data 22...25 (energy total):       0x00 0x00 0x00 0x28 (0.4 kWh)
Data 26...29 (runtime total):      0x00 0x00 0x00 0x0C (12 h)
Data 30...31 (mode):               0x00 0x02 (2: Normal)
Data 32...33 (grid voltage fault): 0x00 0x00 (0.0 V)
Data 34...35 (grid freq. fault)    0x00 0x00 (0.00 Hz)
Data 36...37 (dc injection fault): 0x00 0x00 (0 mA)
Data 38...39 (temperature fault):  0x00 0x00
Data 40...41 (pv1 voltage fault):  0x00 0x00
Data 42...43 (pv2 voltage fault):  0x00 0x00
Data 44...45 (gfc fault):          0x00 0x00
Data 46...49 (error message):      0x00 0x00 0x00 0x00
Data 50...52 (unknown):            0x00 0x01
Data 53...54 (unkown g2):          0x00 0x8A
Data 55...56 (unkown g3):          0x00 0xDE
```

syssi, 2022-07-27:
Could you provide the exact model name and firmware version numbers of your device? I would like to add something like this to the lst of supported devices:

```
SolaX X1 Mini X1-2.0-S-D(L) (master version 1.08, manager version 1.07)
```

syssi, 2022-07-27:
Next wish: Could you increase the log level of your ESPHome node to `VERY_VERBOSE` and provide a complete status response of your device? The log message should start with `RX`:

```
[15:42:33][VV][modbus_solax:083]: RX <- AA.55.00.0A.01.00.11.82.32.00.21.00.02.07.DB.00.00.00.1D.00.00.00.18.09.56.13.81.02.2C.FF.FF.00.00.5D.AF.00.00.10.50.00.02.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.07.96 (61)
```

syssi, 2022-07-27:
Please change the `external_components` source to:

```
external_components:
  - source: github://syssi/esphome-modbus-solax-x1@add-x1-mini-g3-support
    refresh: 0s
```

reflash your device and give it a try!

neujbit, 2022-07-27:
So fast ,seems to be working Thank you,
Model Solax X1-0.6-S-D(L)  Master 1.08  Manager 1.07 

[08:51:52][VV][scheduler:196]: Running interval 'update' with interval=5000 last_execution=4294966968 (now=4678)
[08:51:52][VV][modbus_solax:227]: TX -> AA.55.01.00.00.0A.11.02.00.01.1D (11)
[08:51:52][VV][uart.arduino_esp32:150]:     Flushing...
[08:51:57][VV][scheduler:196]: Running interval 'update' with interval=5000 last_execution=4672 (now=9672)
[08:51:57][VV][modbus_solax:227]: TX -> AA.55.01.00.00.0A.11.02.00.01.1D (11)
[08:51:57][VV][uart.arduino_esp32:150]:     Flushing...
[08:51:57][VV][modbus_solax:089]: RX <- AA.55.00.0A.01.00.11.82.38.00.1A.00.03.04.0C.00.00.00.19.00.00.00.0B.08.FC.13.8A.00.F8.FF.FF.00.00.00.2B.00.00.00.0D.00.02.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00.8A.00.DE.08.5F (67)
[08:51:57][V][sensor:074]: 'solax-x1 energy total': Received new state 4.300000
[08:51:57][D][sensor:125]: 'solax-x1 energy total': Sending state 4.30000 kWh with 3 decimals of accuracy
[08:51:57][V][mqtt:415]: Publish(topic='solax-x1-wechselrichter/sensor/solax-x1_energy_total/state' payload='4.300' retain=1)
[08:51:57][V][sensor:074]: 'solax-x1 runtime total': Received new state 13.000000
[08:51:57][D][sensor:125]: 'solax-x1 runtime total': Sending state 13.00000 h with 0 decimals of accuracy
[08:51:57][V][mqtt:415]: Publish(topic='solax-x1-wechselrichter/sensor/solax-x1_runtime_total/state' payload='13' retain=1)
[08:51:57][V][sensor:074]: 'solax-x1 temperature': Received new state 26.000000
[08:51:57][D][sensor:125]: 'solax-x1 temperature': Sending state 26.00000 °C with 1 decimals of accuracy
[08:51:57][V][mqtt:415]: Publish(topic='solax-x1-wechselrichter/sensor/solax-x1_temperature/state' payload='26.0' retain=1)
[08:51:57][V][sensor:074]: 'solax-x1 energy today': Received new state 0.300000
[08:51:57][D][sensor:125]: 'solax-x1 energy today': Sending state 0.30000 kWh with 3 decimals of accuracy
[08:51:57][V][mqtt:415]: Publish(topic='solax-x1-wechselrichter/sensor/solax-x1_energy_today/state' payload='0.300' retain=1)
[08:51:57][V][sensor:074]: 'solax-x1 dc1 voltage': Received new state 103.599998
[08:51:57][D][sensor:125]: 'solax-x1 dc1 voltage': Sending state 103.60000 V with 1 decimals of accuracy
[08:51:57][V][mqtt:415]: Publish(topic='solax-x1-wechselrichter/sensor/solax-x1_dc1_voltage/state' payload='103.6' retain=1)
[08:51:57][V][sensor:074]: 'solax-x1 dc2 voltage': Received new state 0.000000
[08:51:57][D][sensor:125]: 'solax-x1 dc2 voltage': Sending state 0.00000 V with 1 decimals of accuracy
[08:51:57][V][mqtt:415]: Publish(topic='solax-x1-wechselrichter/sensor/solax-x1_dc2_voltage/state' payload='0.0' retain=1)
[08:51:57][V][sensor:074]: 'solax-x1 dc1 current': Received new state 2.500000
[08:51:57][D][sensor:125]: 'solax-x1 dc1 current': Sending state 2.50000 A with 1 decimals of accuracy
[08:51:57][V][mqtt:415]: Publish(topic='solax-x1-wechselrichter/sensor/solax-x1_dc1_current/state' payload='2.5' retain=1)
[08:51:57][V][sensor:074]: 'solax-x1 dc2 current': Received new state 0.000000
[08:51:57][D][sensor:125]: 'solax-x1 dc2 current': Sending state 0.00000 A with 1 decimals of accuracy
[08:51:57][V][mqtt:415]: Publish(topic='solax-x1-wechselrichter/sensor/solax-x1_dc2_current/state' payload='0.0' retain=1)
[08:51:57][V][sensor:074]: 'solax-x1 ac current': Received new state 1.100000
[08:51:57][D][sensor:125]: 'solax-x1 ac current': Sending state 1.10000 A with 2 decimals of accuracy
[08:51:57][V][mqtt:415]: Publish(topic='solax-x1-wechselrichter/sensor/solax-x1_ac_current/state' payload='1.10' retain=1)
[08:51:57][V][sensor:074]: 'solax-x1 ac voltage': Received new state 230.000000
[08:51:57][D][sensor:125]: 'solax-x1 ac voltage': Sending state 230.00000 V with 1 decimals of accuracy
[08:51:57][V][mqtt:415]: Publish(topic='solax-x1-wechselrichter/sensor/solax-x1_ac_voltage/state' payload='230.0' retain=1)
[08:51:57][V][sensor:074]: 'solax-x1 ac frequency': Received new state 50.020000
[08:51:57][D][sensor:125]: 'solax-x1 ac frequency': Sending state 50.02000 Hz with 1 decimals of accuracy
[08:51:57][V][mqtt:415]: Publish(topic='solax-x1-wechselrichter/sensor/solax-x1_ac_frequency/state' payload='50.0' retain=1)
[08:51:57][V][sensor:074]: 'solax-x1 ac power': Received new state 248.000000
[08:51:57][D][sensor:125]: 'solax-x1 ac power': Sending state 248.00000 W with 0 decimals of accuracy
[08:51:57][V][mqtt:415]: Publish(topic='solax-x1-wechselrichter/sensor/solax-x1_ac_power/state' payload='248' retain=1)
[08:51:57][V][sensor:074]: 'solax-x1 mode': Received new state 2.000000
[08:51:57][D][sensor:125]: 'solax-x1 mode': Sending state 2.00000  with 0 decimals of accuracy
[08:51:57][V][mqtt:415]: Publish(topic='solax-x1-wechselrichter/sensor/solax-x1_mode/state' payload='2' retain=1)
[08:51:57][V][sensor:074]: 'solax-x1 error bits': Received new state 0.000000
[08:51:57][D][sensor:125]: 'solax-x1 error bits': Sending state 0.00000  with 0 decimals of accuracy
[08:51:57][V][mqtt:415]: Publish(topic='solax-x1-wechselrichter/sensor/solax-x1_error_bits/state' payload='0' retain=1)
[08:51:57][V][text_sensor:016]: 'solax-x1 errors': Received new state 
[08:51:57][D][text_sensor:067]: 'solax-x1 errors': Sending state ''
[08:51:57][V][mqtt:415]: Publish(topic='solax-x1-wechselrichter/sensor/solax-x1_errors/state' payload='' retain=1)
[08:51:57][V][text_sensor:016]: 'solax-x1 mode name': Received new state Normal
[08:51:57][D][text_sensor:067]: 'solax-x1 mode name': Sending state 'Normal'
[08:51:57][V][mqtt:415]: Publish(topic='solax-x1-wechselrichter/sensor/solax-x1_mode_name/state' payload='Normal' retain=1)
[08:51:57][V][component:199]: Component modbus_solax took a long time for an operation (0.10 s).
[08:51:57][V][component:200]: Components should block for at most 20-30ms.

syssi, 2022-07-27:
Perfect! I will merge the feature branch to `main` in a few seconds. Please change your `external_components` source back to `main`.

syssi, 2022-07-27:
Please feel free to create new issues if you miss something or if you see room for improvements.

PR #23 (2022-07-27): Add Solax X1 Mini G3 support
Closes: #22

```
external_components:
  - source: github://syssi/esphome-modbus-solax-x1@add-x1-mini-g3-support
    refresh: 0s
```
closed 2022-07-27 by syssi