syssi/esphome-solax-x1-mini #32: Meter functionality

leramleram, 2023-01-04:
Hi There.
I stumbled upon this project, and I´m curious if there is some "external meter functionality" planned?

As you know Solax inverter support external power meters in order to be able to handle export control.
I have a larger PV set than allowed, thus I want to limit the power which gets exportet to the grid.

On the other hand I do not want to always limit the inverter output power because I want to use full PV power when some consumer in my House demands it.

As I have already a power meter installed (no modbus/not supported by Solax) and it´s values are available in my home network, it´d be great to have the possibility to feed these values to the Solax inverter.
Then I would be able to perfectly limit the exported power to the grid.

syssi, 2023-01-04:
There are no plans yet. Did you see this issue? https://github.com/syssi/esphome-modbus-solax-x1/issues/20

The details could be used to emulate a SDM.

leramleram, 2023-01-04:
Hmm... good point. I´ve seen this post somewhere before, but did not reference it to this project... ;-)
I´ll have a look if that could help.
I opened this issue because I think I´m not the only one in such situation.
Of course, having an all-in-one solution which could be tailored through a config would be ultimate.

Thank you for pointing out the direction.

syssi, 2023-01-04:
If you provide a lot of testing we could try to implement a simple custom component to control the limiter of your inverter. Which Solax model do you own? Do you use this project to retrieve measurements from the device?

leramleram, 2023-01-04:
Sounds pretty cool. Testing would be no problem.
I own a Solax X1 mini 1.5.

I have a small wifi gateway at the moment which cyclically connects to the inverter (dongle´s) WIFI.
It fetches the data, and pushes it via MQTT to my home automation.

For future I want to use only your component to send/receive data to/from the inverter.
Actually I´m setting up the hw connection via RS485. (Alltough not succesfull yet, just figuring out what´s wrong with my setup) :)

syssi, 2023-01-04:
Alright! Let's get your RS485 connection running as first step because the connection is required to control the limiter of your inverter too.

leramleram, 2023-01-15:
Still working to get my inverter connected. 

First I had issues with the RS485 adapter, then I checked everything with the oscilloscope.
I can see the request´s going out on the RS485 bus, but the inverter never responds.

Unfortunately I have no option to enable Modbus nor have an option to set the inverters Modbus adress in the inverter´s settings. (others seem to have this option..)
When I try to activate it via the Solax cloud I always get an error when storing the settings for Modbus.... :(
Not sure if thats caused by an older firmware.

I wrote an email to the Solax support.

syssi, 2023-01-15:
If the solax support provides a firmware bundle please let me know the different filenames / version numbers.

leramleram, 2023-01-20:
Received a rather sad message from support this week...
Apparently my inverter is a G1 version which does not support ModbusRTU...

They sent me a protocol description for single phase inverters (v1.8)
Not sure if that helps...

Maybe also same effect on https://github.com/syssi/esphome-modbus-solax-x1/issues/33 ?

syssi, 2023-01-20:
Please be care about the wording: The "SolaX X1 Mini" doesn't support ModbusRTU and this external component doesn't implement the ModbusRTU protocol. It implements a derivation of Modbus using RS485 a transport.

Could you provide a photo of your type label of your inverter? Could you ask the support to provide a firmware update for your device? I would like to use these information to get an idea you have using the same inverter like me or not.

leramleram, 2023-01-20:
Ok sorry for misinterpreting the protocols, I´m quite new with modbus.

The support guy told me that there is RS485 connection available but unmaintained by solax because it is an old gerneration of the inverter.

After I asked if I could upgrade the unit he sent me the protocol.
He also suggested to only use it for meter connection.

Here´s the label on the inverter.

[image]

Is it maybe only a different protocol? Do you see a chance to get it working when I would send you the protocol?

syssi, 2023-01-20:
Feel free to send me the protocol (basti@linkt.de). I will have a look!

syssi, 2023-01-20:
Your label like looks like mine (ignoring the 600W vs. 1500W).

leramleram, 2023-01-28:
So, I ordered a HW0519 adapter to connect to the inverter. Now it works like a charme! 
Connection is up and running. :)

Now we could proceed to get meter functionality working.

syssi, 2023-01-28:
Perfect! As next step we will try pro transmit some limiting instructions to the inverter. I will try to prepare a YAML configuration but it will take some time. Could you check your inverter settings (on screen display) in the meantime. Is there a menu item called "Meter/CT enable" or something like that?

syssi, 2023-01-28:
Could you provide your current YAML config as working basis?

syssi, 2023-01-28:
1. Please flash this test configuration to your ESP: https://raw.githubusercontent.com/syssi/esphome-modbus-solax-x1/export-control/tests/esp8266-dummy-receiver.yaml
2. `Enable` the `Export control` mode at your inverter settings
3. Please provide the log of the ESPHome node. It should contain incoming raw traffic. I expected three different frames:

```
0x01, 0x03, 0x00, 0x0b, 0x00, 0x01, 0xf5, 0xc8 # Handshake
0x01, 0x03, 0x00, 0x0e, 0x00, 0x01, 0xe5, 0xc9 # Request power consumption
0x01, 0x03, 0x00, 0x08, 0x00, 0x04, 0xc5, 0xcb # Request total energy
```

If this is the case I will implement a component which is able to respond to these requests.

leramleram, 2023-01-29:
Great! :)

Firstly mentioned, I noticed some strange behaviour yesterday which I can not explain yet.
From time to time all values from the inverter drop to zero. (looks like regulary intervalls)
Not sure where this comes from. I´ll try to get a debug log to pin down this issue.
Do you have an idea about that?

[image]

> Perfect! As next step we will try to transmit some limiting instructions to the inverter. I will try to prepare a YAML configuration but it will take some time. Could you check your inverter settings (on screen display) in the meantime. Is there a menu item called "Meter/CT enable" or something like that?

Yes I have the "Meter/CT enable" option.

> Could you provide your current YAML config as working basis?

Here it is.
`substitutions:
  name: solar-powermeter
  device_description: "Monitor a Solax X1 mini via RS485"
  external_components_source: github://syssi/esphome-modbus-solax-x1@main
  tx_pin: GPIO17
  rx_pin: GPIO16

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-modbus-solax-x1"
    version: 1.1.1

esp32:
  board: wemos_d1_mini32

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
#mqtt:
#  broker: !secret mqtt_host
#  username: !secret mqtt_username
#  password: !secret mqtt_password
#  id: mqtt_client

api:

logger:
#  baud_rate: 0
  level: INFO

uart:
  id: uart2
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  #debug:
  #  direction: BOTH

modbus_solax:
  - id: modbus0
    uart_id: uart2

solax_x1:
  modbus_solax_id: modbus0
  serial_number: "3132333435363737363534333231"
  address: 0x0A
  update_interval: 3s

text_sensor:
  - platform: solax_x1
    mode_name:
      name: "${name} mode name"
    errors:
      name: "${name} errors"

sensor:
  - platform: solax_x1
    ac_power:
      name: "${name} ac power"
    energy_today:
      name: "${name} energy today"
    energy_total:
      name: "${name} energy total"
    dc1_voltage:
      name: "${name} dc1 voltage"
    #dc2_voltage:
    #  name: "${name} dc2 voltage"
    dc1_current:
      name: "${name} dc1 current"
    #dc2_current:
    #  name: "${name} dc2 current"
    ac_current:
      name: "${name} ac current"
    ac_voltage:
      name: "${name} ac voltage"
    ac_frequency:
      name: "${name} ac frequency"
    temperature:
      name: "${name} temperature"
    runtime_total:
      name: "${name} runtime total"
    mode:
      name: "${name} mode"
    error_bits:
      name: "${name} error bits"
`

syssi, 2023-01-29:
Please follow [this](https://github.com/syssi/esphome-modbus-solax-x1/issues/32#issuecomment-1407489714) how to and don't forget to change the RX/TX pins to `GPIO16`, `GPIO17`.

To debug the gaps at your chart it would be helpful to add additional charts (PV voltage, PV current) and text sensors as `history` card to your lovelace dashboard. Are the sensor values unavailable from time to time or does the Grid Tied Inverter stops operating? In my case I've a single gap per day because the branch of a tree casts a shadow. This this narrow shadow walks over the entire panel within 30 minutes. Afterwards the generation starts again.

leramleram, 2023-01-29:
OK, looks like u´re right that the inverter stops at such low power.
I´ve also my second way of polling the inverter via WIFI still working, and there I also see these gaps.
Strange anyways because I never observed them before.
(which doesn´t mean that they were there. Maybe inverter restarts at such low power situations)

l try to catch the next gap with enabled logging (should come in in roughly 20 mnutes) just to be sure.

After that I will continue with your new config.

leramleram, 2023-01-29:
OK, one hour of continuous operation without gaps. Seems to be related to the low power situation yesterday.

I´ve flashed your test config and enabled the meter on my inverter.
Inverter complains now about meter fault.

Here are the logs.
[logs_esphomedevice_upload.txt](https://github.com/syssi/esphome-modbus-solax-x1/files/10529512/logs_esphomedevice_upload.txt)

syssi, 2023-01-29:
Looks good! The inverter is trying to talk to the meter:

```
[11:05:04][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[11:05:04][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[11:05:04][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
```

This is the handshake frame. As soon the virtual meter is implemented we will respond to this frame. You can flash your production YAML again. I will need some time to implement the virtual meter component.

leramleram, 2023-01-29:
Sounds good!
Btw: Don´t rush, wheather is shitty anyways at the moment..

Thank you so far! :)

leramleram, 2023-01-29:
One thing ybout the implementation:
The pipeline from measuring of the grid parameters to the meter input of the inverter hast a significant potential to fail at one point.
(Current clamps -> Measuring circuit/device -> Can bus -> Raspi -> Mqtt -> Homeassistant -> Solax inverter)

I´d suggest to introduce some kind of timeout. When no vaild grid information arrives at your component for x seconds it should not answer the requests of the inverter so that it returns into a fault state.
Therefore it shouldn´t  be prone of exporting too much if metering fails somehow.

Do you agree?

syssi, 2023-02-05:
A first draft is ready. It's untested but it could work. Please give it a try:

https://raw.githubusercontent.com/syssi/esphome-modbus-solax-x1/export-control/esp8266-virtual-meter.yaml

leramleram, 2023-02-06:
Wow!… 1300 lines of new code in addition... I surely underestimated the amount of work you had.
Unfortunately I can´t test it until weekend because it´s dark when I come home from work.
Alltough I´m quite curious about it!! :)
Maybe I have one day in homeoffice, we´ll see.

Thank you so far!

leramleram, 2023-02-10:
OK, first test done.

Inverter complains about meter fault.
I have tested it with the dummy meter.
[logs_esphomedevice_upload(3).txt](https://github.com/syssi/esphome-modbus-solax-x1/files/10704521/logs_esphomedevice_upload.3.txt)

syssi, 2023-02-10:
Thanks! I will have a look!

syssi, 2023-02-20:
I've fixed the meter fault issue but the requested power demand is hardcoded to -4551W at the moment. You could give it another try already.

leramleram, 2023-02-22:
I was able to test it for a few minutes this evening.

Meter fault has gone away, the inverter seems fine with the responses from the esp32 now :)
I had only a few watts of solar power, so I couldn´t test the export function.

What I´ve noticed in the Solax app is that grid power is still 0.
Is this intended?

I yet don´t understand how your implementation is supposed to work.
Isn´t it just forwarding grid export power to the inverter and the inverter takes care of limiting export power to the value I´ve set in the inverter for max export?
But what about the "power demand", "max power demand" and "buffer" values?

Here you find some screenshots and the log of today.

[image]

[image]

[logs_esphomedevice_logs(1).txt](https://github.com/syssi/esphome-solax-x1-mini/files/10806748/logs_esphomedevice_logs.1.txt)

Thank you for your effort so far and tell me if there´s something new to test.

syssi, 2023-02-22:
I've updated the branch a few minutes ago and removed all useless features. The new YAML configuration looks like this: https://github.com/syssi/esphome-solax-x1-mini/blob/export-control/esp8266-virtual-meter.yaml

Is your smartmeter available via Home Assistant or MQTT? Could you tell me the entity id oder topic?

syssi, 2023-02-22:
The updated implementation is more like your expectation:

1. The smartmeter power value is passed to the Solax if requested
2. If the smartmeter doesn't provide sensor updated to Solax requests will be no longer answered. This will trigger a "meter fault" error and the solax should shutdown.
3. If the "emergency power off switch" is turned off the Solax requests will be no longer answered. This will trigger a "meter fault" error and the solax should shutdown.

That's all. As long you are using the temperature sensor (fake smartmeter value) the feedback loop is open and the inverter will increase the output power permanently.

leramleram, 2023-02-22:
Ok sounds great! :)
Just wanted to make sure that I don´t get it wrong.

Basically the values from my meter come in via MQTT. Cloud mirror it into homeassistant ut that´s probably only one more edge which can fail.
So if possible to feed the values to the ESP via MQTT I would be good with it.

The topic from the meter for active power is `stat/EZ3/P_Wirk_L1`
Positive values indicate power demand from the grid.
Negative values indicate power export to the grid.

Is that OK for the component?
I could also do some remapping with node red to have a separate channel which only holds export power if needed.

syssi, 2023-02-22:
This fits. I will update the example configuration temporary.

syssi, 2023-02-22:
https://github.com/syssi/esphome-solax-x1-mini/blob/export-control/esp8266-virtual-meter.yaml

Please make sure there is a `secrets.yaml` available providing the required secrets:

```
wifi_ssid: MY_WIFI_SSID
wifi_password: MY_WIFI_PASSWORD

mqtt_host: MY_MQTT_HOST
mqtt_username: MY_MQTT_USERNAME
mqtt_password: MY_MQTT_PASSWORD
```

leramleram, 2023-02-25:
I´ve flashed the new component yesterday evening. Unfortunately it was too dark already.
But today... Bam... Meter fault.
Unfortunately I cannot get logs anymore.
Did something change there?
I mean I can flash the device so it seems to be online but no logs...

[image]

Another thing: does it work when no mqtt user/passwd is set?
(I know, bad idea, but for testing...)

syssi, 2023-02-25:
Please restart the ESP. I assume it crashed or doesn't have power.

If you don't want to use credentials you can remove the complete `username` and `password` line from the YAML (mqtt section).

leramleram, 2023-02-25:
> Please restart the ESP. I assume it crashed or doesn't have power.

Done already a few times.

> If you don't want to use credentials you can remove the complete `username` and `password` line from the YAML (mqtt section).

Also done.

Here´s my config. Is it a problem that I´m on an esp32?
(it worked before on the ESP32)

`substitutions:
  name: solar-powermeter
  device_description: "Control the export of a Solax X1 mini via RS485"
  external_components_source: github://syssi/esphome-modbus-solax-x1@export-control
  tx_pin: GPIO17
  rx_pin: GPIO16

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-modbus-solax-x1"
    version: 1.1.0

esp32:
  board: wemos_d1_mini32

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:

mqtt:
  broker: !secret mqtt_host
  #username:
  #password:
  id: mqtt_client

api:
  reboot_timeout: 0s

logger:
  level: DEBUG

uart:
  id: uart0
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH

solax_meter_modbus:
  - id: modbus0
    uart_id: uart0
#    flow_control_pin: GPIO0

solax_virtual_meter:
  solax_meter_modbus_id: modbus0
  address: 0x01

  # The state of this sensor (instantaneous power in watt) is used as source
  power_id: powermeter0
  power_sensor_inactivity_timeout: 5s
  update_interval: 1s

sensor:
  - id: powermeter0
    internal: true
    platform: mqtt_subscribe
    name: "${name} instantaneous power consumption"
    topic: "stat/EZ3/P_Wirk_L1"
    accuracy_decimals: 2
    unit_of_measurement: W
    device_class: power

  - platform: solax_virtual_meter
    power_demand:
      name: "${name} power demand"

text_sensor:
  - platform: solax_virtual_meter
    operation_mode:
      name: "${name} operation mode"

switch:
  - platform: solax_virtual_meter
    emergency_power_off:
      name: "${name} emergency power off"
      restore_mode: RESTORE_DEFAULT_OFF

`

syssi, 2023-02-25:
Could you take the ESP to you computer and attach the device using USB. You can use `esphome logs yourconfig.yaml` or any other terminal software to have a look at the log output (of the serial interface). If it's a boot loop we should see it here.

leramleram, 2023-02-25:
Thx for that hint.
I appended the port to the broker ip adress... 
[image]

Looks way better now!
[image]

Inverter is fine now with the power fed by the esp32.
Now I also see the grid power in my Solax app.

Unfortunately wheather is overcast right now so I cannot test it yet...

syssi, 2023-02-26:
The most important thing I would like to know: Is the power consumption (passed to the inverter) correctly signed or must we invert the value?

leramleram, 2023-02-26:
The signing looks correct. Alltough I´ve already tried it inverse because the representation in the Solax app is a bit puzzling.

[image]

syssi, 2023-02-26:
Another conversation about the same detail: https://github.com/syssi/esphome-solax-x1-mini/issues/20#issuecomment-1076090488

leramleram, 2023-02-26:
I was able to test the limiting functionality today.
It works as expected!! 👍 👍 👍 

Thank you Sebastian for your great work!!

I tried to include also the reading sensors for the inverter, but the weren´t available with the virtual-meter component.
Is this because it´s still in a separate branch?

syssi, 2023-02-26:
No it's because the RS485 interface can be used to talk to a power meter (solax=master, powermeter=slave) or to retrieve sensor informations (computer=master, solax=slave). It's not possible to mix here unfortunately.

closed 2023-02-26 by syssi

leramleram, 2023-03-01:
That´s a pitty..
Anyways I´m verry happy with the meter functionality now. 

Thanks again for your effort!