syssi/esphome-solax-x1-mini #33: Support for solax X1-1.1-S-D(L)

marti85, 2023-01-05:
Hello,
I have a Solax inverter Model X1-1.1-S-D(L) version Manager.1.07 Master: 1.08 to which I have connected an RS485-to-TTL module and a Wemos D1 mini pin TX (GPIO1) and RX (GPIO3) . Unfortunately I cannot recover the data from my inverter. See logs:
[image]
With my oscilloscope I see signals on the TX / RX pins and I see the TX led of RS485 converter blink periodically.
It looks like my inverter modbus is not active. I tried passwords 2014 and 6868 but I have no option for modbus, only DRM (disable)
Could you help me ?`

marti85, 2023-01-05:
Yes i check the RJ45, I connect the pin 4 (A+), 5 (B-) and 7 (GND) on this module : [https://fr.aliexpress.com/item/1005003498933608.html](https://fr.aliexpress.com/item/1005003498933608.html )
I set the logger to 0 like this :
`logger:
  baud_rate: 0
  level: DEBUG`
maybe i will change the RS485 module...

syssi, 2023-01-05:
Please double-check the RJ45 pinout. Don't trust the colors of your cable because of the different standards. I assume it's a connection/RS485 module issue.

closed 2023-01-28 by syssi