syssi/esphome-solax-x1-mini #38: Temperature reading wrong when temperature is negative

leramleram, 2023-02-10:
The temperature reading of the inverter seems to be a signed value, but the component interprets it as unsigned.
This results in an overflow when Temperature is below 0°C.

[image]

syssi, 2023-02-10:
Good catch! I will prepare a fix!

syssi, 2023-02-10:
I pushed a fix to `main`. Please recompile & flash your device and give it another try while it's still cold. ;-)

closed 2023-02-10 by syssi

leramleram, 2023-02-10:
Too late... Temperature´s above 0°C ;-)
Have to give it a try tomorrow!

[image]

leramleram, 2023-02-11:
That´s a Bingo!
[image]

Temperature is now correct when < 0°C

Thank you!

syssi, 2023-02-11:
Perfect! Thanks for keeping your eyes open! :-)

commit cf79352a (2023-02-10): Fix temperature sensor (Closes: #38) #39

It's a signed integer. Kudos to @leramleram