syssi/esphome-solax-x1-mini #56: Solax X1 Mini with Limiter and SDM72 Smartmeter

dirk99, 2023-06-04:
Hello, 
I looked at the Tread #20/32 but I don't understand what I need to adjust for my solution. I would like to control the Soalx inverter with the power meter. I'm using an SDM72 which is not from Solax but which counts 3 phases. Accordingly, the SDM72 also has consultative registers, as is the case with the SDM230 with the 30013 register. For the SDM72 it would be the register 30053 Total System Power. What does the digestion of the Modbus have to look like? AB from the Sloax parallel with Smartmeter and the esphome? The ESP Home only uses one UART. Does the ESPHOME convert any data between smartmeter ans solax inverter?

Many thanks in advance.

Greetings Dirk

syssi, 2023-06-04:
I will provide some details tomorrow. What's the distance between the SDM and your Solar?

dirk99, 2023-06-04:
The distance is 40-45m

syssi, 2023-06-04:
In this case I would suggest two ESPHome node. One for driving the limiter of the solax using this project and another ESP to retrieve the power consumption from the SDM and publish the measurement via MQTT or native API.

dirk99, 2023-06-05:
The distance is no problem. I suggest one esp node. Between the two components exist cat7 cable.

syssi, 2023-06-05:
Alright. A single ESP setup is fine too. ESPHome supports multiple UARTs on a ESP8266 because of a software serial implementation. There is just one hardware UART available. If you assign any other GPIO to the `uart` component the software serial implementation is used. If you don't trust the software serial implementation please use a ESP32. Here are 3 hardware UARTs accessible.

If you enable the meter mode of the Solax inverter the device will be the modbus **master**. The solax polls periodically a specific slave address. This ESPHome implementation forms a modbus **slave** and is able to react to the Solax queries. This setup requires the first UART.

A second UART is required for the RS485 communication between the SDM and the ESPHome node. On this bus the ESPHome is the modbus **master** and the SDM response to requests as **slave**. 

The final setup should look like this:

```
Solax <-- RS485 module --> ESP32 <--> RS485 module <--> SDM
```

Feel free to ask more questions. If you are new to ESPHome there are a lot of concepts you should know/learn. :-)

dirk99, 2023-06-05:
What is the function of your project esp8266-meter-gateway.yaml?

syssi, 2023-06-05:
Decoding the solax meter request and respond.

syssi, 2023-06-05:
I assume you will remove the MQTT part from this configuration and use the SDM component instead to provide the power consumption sensor entity: https://esphome.io/components/sensor/sdm_meter.html

dirk99, 2023-06-05:
If I wrote the mqtt stat.EZ3.P_Wirk_L1 with the power consumption from smart meter your scrip transmit it to the solax inverter? This should be a easy way for me, because I have all thinks in my iobroker.

syssi, 2023-06-05:
Correct! The implementation does pick up a measurement (float) from the `instantaneous_power_topic` and passes this value to the solax inverter if it's requested (by the solax meter mode).

dirk99, 2023-06-05:
If i connect my Solax inverter to ESPhome ans set continously the MQTT Topic with the consumtion from my smartmeter the log is shown a meter fault 
YAML:
substitutions:
  name: solax-meter-gateway
  device_description: "Control the export of a Solax X1 mini via RS485"
  external_components_source: github://syssi/esphome-modbus-solax-x1@main
  instantaneous_power_topic: "SolaxPowerControl/P_Wirk_L1"
  tx_pin: GPIO17
  rx_pin: GPIO16

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-modbus-solax-x1"
    version: 2.1.0

esp32:
  board: esp32dev
  framework:
    type: arduino

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: <SSID>
  password: <Passphrase>

ota:

logger:
  level: DEBUG

api:
  reboot_timeout: 0s

mqtt:
  broker: <IP>
  port: 2884 
  username: <User>
  password: <Pass>
  id: mqtt_client

uart:
  id: uart_0
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH

solax_meter_modbus:
  - id: modbus0
    uart_id: uart_0
#    flow_control_pin: GPIO0

solax_meter_gateway:
  solax_meter_modbus_id: modbus0
  address: 0x01

  # The state of this sensor (instantaneous power in watt) is used as source
  power_id: powermeter0
  power_sensor_inactivity_timeout: 5s
  update_interval: 1s

sensor:
  - id: powermeter0
    internal: true
[... 95 lines cut ...]
[10:39:17][C][mqtt_subscribe.sensor:031]: MQTT Subscribe 'solax-meter-gateway instantaneous power consumption'
[10:39:17][C][mqtt_subscribe.sensor:031]:   Device Class: 'power'
[10:39:17][C][mqtt_subscribe.sensor:031]:   State Class: ''
[10:39:17][C][mqtt_subscribe.sensor:031]:   Unit of Measurement: 'W'
[10:39:17][C][mqtt_subscribe.sensor:031]:   Accuracy Decimals: 2
[10:39:17][C][mqtt_subscribe.sensor:032]:   Topic: SolaxPowerControl/P_Wirk_L1
[10:39:17][C][mqtt.sensor:027]: MQTT Sensor 'solax-meter-gateway power demand':
[10:39:17][C][mqtt.sensor:031]:   State Topic: 'solax-meter-gateway/sensor/solax-meter-gateway_power_demand/state'
[10:39:17][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'Meter fault'
[10:39:17][D][sensor:094]: 'solax-meter-gateway power demand': Sending state nan W with 0 decimals of accuracy
[10:39:17][W][solax_meter_gateway:021]: No power sensor update received since 5 seconds. Triggering meter fault for safety reasons
[10:39:17][C][mqtt.text_sensor:023]: MQTT Text Sensor 'solax-meter-gateway operation mode':
[10:39:17][C][mqtt.text_sensor:024]:   State Topic: 'solax-meter-gateway/sensor/solax-meter-gateway_operation_mode/state'
[10:39:17][C][mqtt.switch:041]: MQTT Switch 'solax-meter-gateway emergency power off': 
[10:39:17][C][mqtt.switch:042]:   State Topic: 'solax-meter-gateway/switch/solax-meter-gateway_emergency_power_off/state'
[10:39:17][C][mqtt.switch:042]:   Command Topic: 'solax-meter-gateway/switch/solax-meter-gateway_emergency_power_off/command'
[10:39:17][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[10:39:17][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'Meter fault'
[10:39:17][D][sensor:094]: 'solax-meter-gateway power demand': Sending state nan W with 0 decimals of accuracy
[10:39:17][W][solax_meter_gateway:021]: No power sensor update received since 5 seconds. Triggering meter fault for safety reasons
[10:39:17][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[10:39:17][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'Meter fault'
[10:39:17][D][sensor:094]: 'solax-meter-gateway power demand': Sending state nan W with 0 decimals of accuracy
[10:39:18][W][solax_meter_gateway:021]: No power sensor update received since 5 seconds. Triggering meter fault for safety reasons
[10:39:18][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8

syssi, 2023-06-05:
Could you monitor the traffic on topic `SolaxPowerControl/P_Wirk_L1`? It looks like the node doesn't receive any power conumption measurements. If you have access to a linux host please provide the output of:

```
mosquitto_sub -t 'SolaxPowerControl/P_Wirk_L1' -v
```

dirk99, 2023-06-06:
Witch traffic do you want to see? Iobroker has not this functions. Ist a wireshark helpful?

syssi, 2023-06-06:
You could pick any MQTT client software to have a look at your MQTT traffic. It looks like there is no traffic on topic `SolaxPowerControl/P_Wirk_L1` at the moment.

syssi, 2023-06-06:
Ich versuche es mal in Deutsch in der Hoffnung, dass wir uns so flinker verstaendigen koennen: Solange der Komponente `solax_meter_gateway` kein gültiger Wert vom Smartmeter zur Verfügung steht, hat sie keine Information, welche sie an den Solax weitergeben koennte. Das ist meine aktuelle Erklaerung, weshalb gerade nichts an den Solax gesendet wird.

Stattdessen scheint das Problem weiterhin beim Einsammeln des aktuellen Verbrauchswert zu liegen. Ich habe den Eindruck, dass keine Verbindung zu deinem MQTT-Broker aufgenommen werden kann und deshalb auch keine Messwerte vom Topic `SolaxPowerControl/P_Wirk_L1` vorliegen. Ist der Port 2884 für unverschlüsselten Verkehr erreichbar? Ggf. hilft noch ein Wireshark-Mitschnitt um zu verstehen, wie die Kommunikation zwischen ESPHome Node und MQTT Port 2884 aktuell aussieht. Ggf. steht auch im IoBroker-Log etwas, wie mit dem Client umgegangen wird.

dirk99, 2023-06-06:
Der Port 2884 ist unsecured mit User und Passwort zu erreichen. Ich sehe auch mit einem mqtt Client mit den gleichen Einstellungen wie das topic im Rytmus von ca. 2sek geändert wird und auch einen positive bzw. Negativen Wert bekommt mit Nachkommastellen (Float). Ich werde heute Abend mal ein Wireshark Protokoll ziehen. Meine Vermutung liegt eher an der Hardware. Parallel werde ich es mal auf Flow control umbauen. Teile habe ich dafür da. Von: Sebastian Muszynski ***@***.***>Gesendet: Dienstag, Juni 6, 2023 10:48:33 AMAn: syssi/esphome-solax-x1-mini ***@***.***>Cc: dirk99 ***@***.***>; Author ***@***.***>Betreff: Re: [syssi/esphome-solax-x1-mini] Solax X1 Mini with Limiter and SDM72 Smartmeter (Issue #56)
Ich versuche es mal in Deutsch in der Hoffnung, dass wir uns so flinker verstaendigen koennen: Solange der Komponente solax_meter_gateway kein gültiger Wert vom Smartmeter zur Verfügung steht, hat sie keine Information, welche sie an den Solax weitergeben koennte. Das ist meine aktuelle Erklaerung, weshalb gerade nichts an den Solax gesendet wird.
Stattdessen scheint das Problem weiterhin beim Einsammeln des aktuellen Verbrauchswert zu liegen. Ich habe den Eindruck, dass keine Verbindung zu deinem MQTT-Broker aufgenommen werden kann und deshalb auch keine Messwerte vom Topic SolaxPowerControl/P_Wirk_L1 vorliegen. Ist der Port 2884 für unverschlüsselten Verkehr erreichbar? Ggf. hilft noch ein Wireshark-Mitschnitt um zu verstehen, wie die Kommunikation zwischen ESPHome Node und MQTT Port 2884 aktuell aussieht. Ggf. steht auch im IoBroker-Log etwas, wie mit dem Client umgegangen wird.

syssi, 2023-06-06:
```
[10:11:33][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
```

zeigt an, dass eine Nachricht vom Solax herein kommt. Es handelt sich um das Handshake-Frame, welches von der Komponente unmittelbar mit einer ausgehenden Nachricht (`>>>`) beantwortet werden sollte:

https://github.com/syssi/esphome-solax-x1-mini/blob/main/components/solax_meter_gateway/solax_meter_gateway.cpp#L33-L37

Ich habe das Gefuehl, dass in deiner Konfiguration noch etwas nicht stimmt. Egal ob die Hardware funktioniert wie sie soll, haette man die ausgehende Nachricht (`01:03:02:00:00`) im Log sehen muessen. Ich gehe davon aus, dass deine Hardware sauber funktioniert, da du Nachrichten vom Solax empfangen kannst.

Kannst du mir deine YAML-Konfiguration zur Verfügung stellen? Dann kann ich deinen Aufbau nachstellen.

dirk99, 2023-06-06:
anbei meine Config und ein Log. im Log sieht um 15:03 für mich so aus, als wenn das ein paar Daten geschickt werden, aber nach ein paar Meldungen ist Schluß. Kann es daran liegen, dass paralell noch ein Tasmota auf dem Bus die Abfragen am Solax macht? Ist es eigentlich möglich, dass die Smartmeter Konvertierung und parallel die Datenabfrage vom ESPhome per MQTT übermittelt wird, dann könnte ich den Tasmota vom Bus entfernen.
Config YAML

substitutions:
  name: solax-meter-gateway
  device_description: "Control the export of a Solax X1 mini via RS485"
  external_components_source: github://syssi/esphome-modbus-solax-x1@main
  instantaneous_power_topic: "SolaxPowerControl/P_Wirk_L1"
  tx_pin: GPIO4
  rx_pin: GPIO5

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-modbus-solax-x1"
    version: 2.1.0

esp8266:
  board: d1_mini

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: home
  password: <SSID Kennwort>

ota:
  password: "<PW>"

logger:
  level: DEBUG

api:
  encryption:
    key: "<key>"

mqtt:
  broker: 192.168.168.201
  port: 2884 
  username: <mqtt_user>
  password: <mqtt_PW>
  id: mqtt_client

uart:
  id: uart_0
  baud_rate: 9600
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH

solax_meter_modbus:
  - id: modbus0
    uart_id: uart_0
#    flow_control_pin: GPIO0

solax_meter_gateway:
  solax_meter_modbus_id: modbus0
  address: 0x01

  # The state of this sensor (instantaneous power in watt) is used as source
  power_id: powermeter0
  power_sensor_inactivity_timeout: 5s
  update_interval: 1s

sensor:
  - id: powermeter0
[... 216 lines cut ...]
[15:03:05][W][solax_meter_gateway:020]: No power sensor update received since 5 seconds. Triggering meter fault for safety reasons
[15:03:05][D][uart_debug:114]: <<< 01:03:00:0E:00:01:E5:C9
[15:03:05][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'Meter fault'
[15:03:05][D][sensor:093]: 'solax-meter-gateway power demand': Sending state nan W with 0 decimals of accuracy
[15:03:05][W][solax_meter_gateway:020]: No power sensor update received since 5 seconds. Triggering meter fault for safety reasons
[15:03:05][D][uart_debug:114]: <<< 01:03:00:0E:00:01:E5:C9
[15:03:05][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'Meter fault'
[15:03:05][D][sensor:093]: 'solax-meter-gateway power demand': Sending state nan W with 0 decimals of accuracy
[15:03:05][W][solax_meter_gateway:020]: No power sensor update received since 5 seconds. Triggering meter fault for safety reasons
[15:03:06][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[15:03:06][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'Meter fault'
[15:03:06][D][sensor:093]: 'solax-meter-gateway power demand': Sending state nan W with 0 decimals of accuracy
[15:03:06][W][solax_meter_gateway:020]: No power sensor update received since 5 seconds. Triggering meter fault for safety reasons
[15:03:06][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[15:03:06][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'Meter fault'
[15:03:06][D][sensor:093]: 'solax-meter-gateway power demand': Sending state nan W with 0 decimals of accuracy
[15:03:06][W][solax_meter_gateway:020]: No power sensor update received since 5 seconds. Triggering meter fault for safety reasons
[15:03:06][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[15:03:06][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'Meter fault'
[15:03:06][D][sensor:093]: 'solax-meter-gateway power demand': Sending state nan W with 0 decimals of accuracy
[15:03:06][W][solax_meter_gateway:020]: No power sensor update received since 5 seconds. Triggering meter fault for safety reasons
[15:03:06][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[15:03:07][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'Meter fault'
[15:03:07][D][sensor:093]: 'solax-meter-gateway power demand': Sending state nan W with 0 decimals of accuracy
[15:03:07][W][solax_meter_gateway:020]: No power sensor update received since 5 seconds. Triggering meter fault for safety reasons

syssi, 2023-06-06:
Der Teil sieht gut aus:

```
[15:03:00][D][uart_debug:114]: <<< 01:03:00:0E:00:01:E5:C9
[15:03:00][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 474.06000 W with 0 decimals of accuracy
[15:03:00][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[15:03:00][D][uart_debug:114]: >>> 01:03:02:01:DA:38:4F
[15:03:00][D][uart_debug:114]: <<< 01:03:00:0E:00:01:E5:C9
[15:03:00][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 474.06000 W with 0 decimals of accuracy
[15:03:00][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[15:03:00][D][uart_debug:114]: >>> 01:03:02:01:DA:38:4F
[15:03:01][D][uart_debug:114]: <<< 01:03:00:0E:00:01:E5:C9
[15:03:01][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 474.06000 W with 0 decimals of accuracy
[15:03:01][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[15:03:01][D][uart_debug:114]: >>> 01:03:02:01:DA:38:4F
[15:03:01][D][uart_debug:114]: <<< 01:03:00:0E:00:01:E5:C9
[15:03:01][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 474.06000 W with 0 decimals of accuracy
[15:03:01][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[15:03:01][D][uart_debug:114]: >>> 01:03:02:01:DA:38:4F
[15:03:01][D][uart_debug:114]: <<< 01:03:00:0E:00:01:E5:C9
[15:03:01][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 474.06000 W with 0 decimals of accuracy
[15:03:01][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[15:03:01][D][uart_debug:114]: >>> 01:03:02:01:DA:38:4F
[15:03:01][D][uart_debug:114]: <<< 01:03:00:0E:00:01:E5:C9
[15:03:01][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 474.06000 W with 0 decimals of accuracy
[15:03:01][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[15:03:02][D][uart_debug:114]: >>> 01:03:02:01:DA:38:4F
```

In diesem Zeitraum kannte die Steuerung einen aktuellen Verbrauch (474W) vom Stromzaehler und hat den Inverter aufgefordert diese Leistung zu produzieren.

Ab hier wurde etwas seltsames empfangen:

```
[15:03:02][D][sensor:093]: 'solax-meter-gateway power demand': Sending state nan W with 0 decimals of accuracy
```

`nan` bedeutet in diesem Kontext "not a number". In anderen Worten: Der Wert konnte nicht sauber in eine Fließkommazahl konvertiert werden. Wir sollten das Log-Level der MQTT-Komponente anheben, so dass wir die rohen Daten auf dem MQTT-Topic im Log sehen können. Ich vermute, dass die Implementierung alle Viere von sich streckt, sobald der Wert negativ wird.

syssi, 2023-06-06:
Doofe Frage aber wie häufig wird der Stromverbrauch auf dem Topic gemeldet? Einmal pro Sekunde oder seltener?

syssi, 2023-06-06:
Du kannst mal versuchen diesen Parameter `power_sensor_inactivity_timeout` anzuheben auf 30 oder 60s. Faellt die Regelung dann seltener / nicht mehr in den "meter fault"-Modus aufgrund fehlender Daten vom Smartmeter?

dirk99, 2023-06-07:
Ich habe mittlerweile das Update vom Smartmeter auf 1sek gestellt und auch das Timeout auf 30 bzw. 60 sek. gestellt. Das Ergebnis bleibt, 5 Datensäte und dann ist Schluß. Dann kommt wieder meter_fault

syssi, 2023-06-07:
Dann habe ich bei der Implementierung einen Fehler gemacht. :-) Ich schau es mir an.

syssi, 2023-06-07:
Ich habe eine kleine Änderung gepusht um etwas gesprächiger im Log-Level `VERY_VERBOSE` zu werden:

https://github.com/syssi/esphome-solax-x1-mini/commit/da853e72442b354d9d8b095e31bb9be3bc473048

Meine Hoffnung ist so der Ursache auf die Schliche zu kommen. Einen Fehler im Code konnte ich bisher nicht ausfindig machen.

syssi, 2023-06-07:
Bitte einfach den ESP noch einmal flashen. Die zusätzlichen Meldungen im Log sollten daraufhin sofort erscheinen.

syssi, 2023-06-07:
```
[11:52:25][W][api.connection:075]: 192.168.168.201: Socket operation failed: BAD_INDICATOR errno=11
```

Hier geht es um die native API, welche normalerweise nur von Home Assistant oder ESPHome selbst genutzt wird. Gibt es auf dem Host 192.168.168.201 etwas in der Richtung? Der `BAD_INDICATOR`-Fehler tritt in der Regel auf, wenn man einen API-Schlüssel verwendet der nicht zur YAML-Konfiguration passt. Vermutlich kannst du diese Zeile ignorieren.

syssi, 2023-06-07:
```
[17:59:08][I][app:102]: ESPHome version 2023.5.5 compiled on Jun 6 2023, 17:54:47
```

Kannst du die Node einmal neu flashen, so dass die Änderung von heute Morgen mit auf den ESP kommt?

dirk99, 2023-06-07:
Welche Version muss denn angezeigt werden? Ich hatte ihn neu geflashtMit freundlichen GrüßenDirk HowoldMobil ***@***.***

syssi, 2023-06-07:
> Was mir aber aufgefallen ist, sobald ich den Solax auf Export Control "Meter" stelle und die Export Controll Value auf "20000" stehen habe liefert der Inverter kein Strom mehr. (AC power 0W). Wenn ich es richtig verstanden habe sollte bei 20000 doch keine Begrenzung statt finden oder?

Ich kenne dazu nur diese Erklaerung, welche aber nicht zu deiner Beobachtung passt:

https://github.com/syssi/esphome-solax-x1-mini/issues/20#issuecomment-1086832787

Hier wird beschrieben, dass man über diesen Wert den Export an das Stromnetz steuert. Steht die Einstellung auf "0 Watt", dann wird versucht keine Energie an den Netzbetreiber zu exportieren. Setzt du es auf "20k" haette ich erwartet, dass im worst-case 20kW in Netz rückgespeist werden.

syssi, 2023-06-07:
> Welche Version muss denn angezeigt werden? Ich hatte ihn neu geflasht

An dem Zeitstempel (compile time) aus dem Log habe ich nur abgelesehen, dass du zuletzt gestern um 17:54 Uhr geflasht hast:

> [17:59:08][I][app:102]: ESPHome version 2023.5.5 compiled on Jun 6 2023, 17:54:47

dirk99, 2023-06-07:
Genau so habe ich die Doku verstanden... HmmmMit freundlichen GrüßenDirk HowoldMobil ***@***.***

dirk99, 2023-06-08:
ich habe es noch einmal heute verifiziert. Scheinbar liest der ESPhome nur einmal den MQTT Wert und gibt diesen dann immer wieder an den Solax Inverter weiter. Ich sehe auch, wie dieser in der Solax Cloud nach dem nächsten Update ankommt. Kannst Du bitte mal das Logging erweitern und jedes mal den MQTT Wert ausgeben. Im ioBroker kann ich sehen. das dieser Wert sich sekündlich ändert. Wenn ich den ESPhome neu starte, kommt einmalig ein neuer Wert. Mit dem Thema dass der Inverten dann auf 0W Produktion geht, werde ich mich an den Solaxsupport wenden, wenn es soweit läuft. Danke schonmal vorab für Deinen SUPER SUPPORT...

syssi, 2023-06-09:
Ich habe folgendes Log-Level gesetzt:

```
logger:
  level: VERY_VERBOSE
  logs:
    scheduler: DEBUG
```

Und die `esp8266-meter-gateway.yaml` daraufhin auf einen ESP8266 geflasht. Daraufhin bucht sich der ESP auf dem MQTT-Broker ein und meldet folgendes:

```
[12:57:50][VV][solax_meter_gateway:089]: New power demand received (-406.53). Resetting inactivity timeout (35626)
[12:57:51][V][sensor:043]: 'solax-meter-gateway instantaneous power consumption': Received new state -408.170013
[12:57:51][D][sensor:093]: 'solax-meter-gateway instantaneous power consumption': Sending state -408.17001 W with 2 decimals of accuracy
[12:57:51][VV][solax_meter_gateway:089]: New power demand received (-408.17). Resetting inactivity timeout (36630)
[12:57:52][V][sensor:043]: 'solax-meter-gateway instantaneous power consumption': Received new state -410.359985
[12:57:52][D][sensor:093]: 'solax-meter-gateway instantaneous power consumption': Sending state -410.35999 W with 2 decimals of accuracy
[12:57:52][VV][solax_meter_gateway:089]: New power demand received (-410.36). Resetting inactivity timeout (37626)
[12:57:53][V][sensor:043]: 'solax-meter-gateway instantaneous power consumption': Received new state -409.049988
[12:57:53][D][sensor:093]: 'solax-meter-gateway instantaneous power consumption': Sending state -409.04999 W with 2 decimals of accuracy
[12:57:53][VV][solax_meter_gateway:089]: New power demand received (-409.05). Resetting inactivity timeout (38633)
[12:57:54][V][sensor:043]: 'solax-meter-gateway instantaneous power consumption': Received new state -407.410004
[12:57:54][D][sensor:093]: 'solax-meter-gateway instantaneous power consumption': Sending state -407.41000 W with 2 decimals of accuracy
[12:57:54][VV][solax_meter_gateway:089]: New power demand received (-407.41). Resetting inactivity timeout (39643)
[12:57:55][V][sensor:043]: 'solax-meter-gateway instantaneous power consumption': Received new state -409.750000
[12:57:55][D][sensor:093]: 'solax-meter-gateway instantaneous power consumption': Sending state -409.75000 W with 2 decimals of accuracy
[12:57:55][VV][solax_meter_gateway:089]: New power demand received (-409.75). Resetting inactivity timeout (40629)
[12:57:56][V][sensor:043]: 'solax-meter-gateway instantaneous power consumption': Received new state -411.790009
[12:57:56][D][sensor:093]: 'solax-meter-gateway instantaneous power consumption': Sending state -411.79001 W with 2 decimals of accuracy
```

Diese Meldungen wuerde ich auch gerne in deinem Logbuch sehen. Insbesondere die Zeile

```
[12:57:55][VV][solax_meter_gateway:089]: New power demand received (-409.75). Resetting inactivity timeout (40629)
```

Als naechstes habe ich einen Solax-Inverter imitiert mit der folgenden Konfiguration:

```
tests/esp8266-query-sdm230.yaml
```

Diese habe ich auf einen zweiten ESP geflasht und die beiden ESPs mit gekreuzten Datenleitungen verbunden. Im Logbuch sollte man nun auch die Anfragen des Fake-Solax sehen und die Komponente antwortet:

```
[13:05:10][V][sensor:043]: 'solax-meter-gateway instantaneous power consumption': Received new state 832.460022
[13:05:10][D][sensor:093]: 'solax-meter-gateway instantaneous power consumption': Sending state 832.46002 W with 2 decimals of accuracy
[13:05:10][VV][solax_meter_gateway:089]: New power demand received (832.46). Resetting inactivity timeout (475639)
[13:05:10][VV][solax_meter_modbus:069]: RX <- 01.03.00.0B.00.01.F5.C8 (8)
[13:05:10][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[13:05:10][VV][uart.arduino_esp8266:180]:     Flushing...
[13:05:10][V][solax_meter_modbus:181]: SolaxMeterModbus write raw: 01.03.02.00.00 (5)
[13:05:10][D][uart_debug:114]: >>> 01:03:02:00:00:B8:44
[13:05:11][V][sensor:043]: 'solax-meter-gateway instantaneous power consumption': Received new state 829.619995
[13:05:11][D][sensor:093]: 'solax-meter-gateway instantaneous power consumption': Sending state 829.62000 W with 2 decimals of accuracy
[13:05:11][VV][solax_meter_gateway:089]: New power demand received (829.62). Resetting inactivity timeout (476642)
[13:05:11][VV][solax_meter_modbus:069]: RX <- 01.03.00.0E.00.01.E5.C9 (8)
[13:05:11][D][uart_debug:114]: <<< 01:03:00:0E:00:01:E5:C9
[13:05:11][VV][uart.arduino_esp8266:180]:     Flushing...
[13:05:11][V][solax_meter_modbus:127]: SolaxMeterModbus write: 01.03.02.03.3D.79.65 (7)
[13:05:11][V][sensor:043]: 'solax-meter-gateway power demand': Received new state 829.619995
[13:05:11][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 829.62000 W with 0 decimals of accuracy
[13:05:11][V][mqtt:422]: Publish(topic='solax-meter-gateway/sensor/solax-meter-gateway_power_demand/state' payload='830' retain=1)
[13:05:11][V][text_sensor:013]: 'solax-meter-gateway operation mode': Received new state On
[13:05:11][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[13:05:11][V][mqtt:422]: Publish(topic='solax-meter-gateway/sensor/solax-meter-gateway_operation_mode/state' payload='On' retain=1)
[13:05:11][V][component:204]: Component solax_meter_modbus took a long time for an operation (0.09 s).
[13:05:11][V][component:205]: Components should block for at most 20-30ms.
[... 6 lines cut ...]

Guckt man ins Logbuch des Fake-Solax findet man hier den reinen UART/RS485-Verkehr:

```
[13:06:15][D][uart_debug:114]: >>> 01:03:00:0B:00:01:F5:C8    # Solax handshake request
[13:06:15][D][uart_debug:114]: <<< 01:03:02:00:00:B8:44       # Gateway handshake response
[... cut ...]
[13:06:26][D][uart_debug:114]: >>> 01:03:00:0E:00:01:E5:C9
[13:06:26][D][uart_debug:114]: <<< 01:03:02:03:51:79:48
[13:06:27][D][uart_debug:114]: >>> 01:03:00:08:00:04:C5:CB
[13:06:27][D][uart_debug:114]: <<< 01:03:08:00:00:00:00:00:00:00:00:95:D7
```

Im Testbett arbeitet die Implementierung, wie sie soll. Wir sollten uns die MQTT-Server-Konfiguration deines ioBrokers einmal genauer ansehen. Ich arbeite mit einer Mosquitto-Instanz als MQTT-Broker, von welcher ich weiss, dass sie sehr gut mit ESPHome harmoniert.

syssi, 2023-06-09:
Kannst du deine MQTT-Konfiguration einmal mit dieser hier vergleichen bzw. ein solches Screenshot zur Verfügung stellen? https://github.com/syssi/esphome-jk-bms/issues/295

dirk99, 2023-06-09:
Es läuft, Danke für Dein Super Support. Es war ein Schicht 8 Fehler bei mir im MQTT Adapter. Es lag an Maske zum Bekanntgeben eigener States. Ich hatte mit einem MQTT Client die Werte gepublished. Somit hat ein weiter Client auch feinedie Werte Angezeigt. aber der Broker hat sie nicht gepublished. Jetzt habe ich noch eine Frage. Kann der gleiche ESPhome parallel auch alle Status Werte periodisch vom Solax Inverter abrufen und via MQTT publishen?

syssi, 2023-06-09:
> Kann der gleiche ESPhome parallel auch alle Status Werte periodisch vom Solax Inverter abrufen und via MQTT publishen?

Leider nein. Das ist aber keine Limitierung des ESPs, sondern eine doofe Limitierung des Solax-Inverters. Im Modbus-Modus ist der Inverter ein Slave auf dem Bus und die ESPHome-Implementierung nimmt die Rolle eine Masters ein und befragt den Solax periodisch nach neuen Messdaten.

Im "Meter Modus" ist der Solax der Master und befragt periodisch das Smartmeter / den SDM oder in unserem Fall das Gateway nach Messwerten. Wir duerfen auf diesem Bus nur antworten, wenn der Master etwas fragt. Selbst wenn wir zusaetzliche Requests fuer Messdaten auf den Bus schicken wuerden, so wuerde der Inverter nicht darauf reagieren.

Moechtest du den Inverter also ueberwachen, dann geht das in deinem Setup nun nicht mehr über den RS485-Anschluss, der ist nun fuer einen anderen Zweck belegt.

closed 2023-06-10 by syssi