syssi/esphome-solax-x1-mini #58: Meter fault (solax-meter-gateway)

dirk99, 2023-07-18:
Hallo,
nachdem ich endlich dazu gekommen bin das gateway live in betrieb zu nehmen, habe ich festgestellt, dass der Umrichter immer auf "Meter fault" stehen bleibt.
Wenn ich das Log richtig interpretiere, bekoomt das Gateway die Werte vom Meter und kommuniziert auch mit dem Solax Umrichter. Warum geht der Umrichter nicht in die Produktion. 
Solax Umrichter ist unter Erweitert - Einspeisebegrenzung auf Meter gestellt und in der Begrenzung auf 20000 für unbegrenzt, oder 0 auf keine Einspeisung gestellt. Der Umrichter bleibt aber mit fehler ausser Betrieb.

Log 
INFO ESPHome 2023.5.5
INFO Reading configuration /opt/iobroker/iobroker-data/esphome.0/solax-meter-gateway-esp32.yaml...
INFO Updating https://github.com/syssi/esphome-modbus-solax-x1.git@main
INFO Starting log output from solax-meter-gateway-esp32.local using esphome API
INFO Successfully connected to solax-meter-gateway-esp32.local
[17:53:58][I][app:102]: ESPHome version 2023.5.5 compiled on Jun 14 2023, 11:06:16
[17:53:58][I][app:104]: Project syssi.esphome-modbus-solax-x1 version 2.1.0
[17:53:58][C][wifi:505]: WiFi:
[17:53:58][C][wifi:363]:   Local MAC: C4:4F:33:6A:XX:XX
[17:53:58][C][wifi:364]:   SSID: 'xxxx'[redacted]
[17:53:58][C][wifi:365]:   IP Address: x.x.x.x
[17:53:58][C][wifi:367]:   BSSID: B4:FB:E4:74:xx:xx[redacted]
[17:53:58][C][wifi:368]:   Hostname: 'solax-meter-gateway-esp32'
[17:53:58][C][wifi:370]:   Signal strength: -71 dB ▂▄▆█
[17:53:58][C][wifi:374]:   Channel: 11
[17:53:58][C][wifi:375]:   Subnet: 255.255.255.0
[17:53:58][C][wifi:376]:   Gateway: x.x.x.x
[17:53:58][C][wifi:377]:   DNS1: x.x.x.x
[17:53:58][C][wifi:378]:   DNS2: 8.8.8.8
[17:53:58][D][sensor:094]: 'solax-meter-gateway-esp32 instantaneous power consumption': Sending state 499.64001 W with 2 decimals of accuracy
[17:53:58][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[17:53:58][C][logger:301]: Logger:
[17:53:58][C][logger:302]:   Level: DEBUG
[17:53:58][C][logger:303]:   Log Baud Rate: 115200
[17:53:58][C][logger:305]:   Hardware UART: UART0
[17:53:58][C][logger:309]:   Level for 'scheduler': DEBUG
[17:53:58][C][uart.arduino_esp32:124]: UART Bus 1:
[17:53:58][C][uart.arduino_esp32:125]:   TX Pin: GPIO17
[17:53:58][C][uart.arduino_esp32:126]:   RX Pin: GPIO16
[17:53:58][C][uart.arduino_esp32:128]:   RX Buffer Size: 256
[17:53:58][C][uart.arduino_esp32:130]:   Baud Rate: 9600 baud
[17:53:58][C][uart.arduino_esp32:131]:   Data Bits: 8
[17:53:58][C][uart.arduino_esp32:132]:   Parity: NONE
[17:53:58][C][uart.arduino_esp32:133]:   Stop bits: 1
[17:53:58][D][uart_debug:114]: >>> 01:03:02:00:00:B8:44
[17:53:58][C][solax_meter_modbus:096]: SolaxMeterModbus:
[17:53:58][C][solax_meter_gateway:095]: SolaxMeterGateway:
[17:53:58][C][solax_meter_gateway:096]:   Address: 0x01
[17:53:58][C][solax_meter_gateway:097]:   Power Demand 'solax-meter-gateway-esp32 power demand'
[17:53:58][C][solax_meter_gateway:097]:     Device Class: 'power'
[17:53:58][C][solax_meter_gateway:097]:     State Class: ''
[17:53:58][C][solax_meter_gateway:097]:     Unit of Measurement: 'W'
[17:53:58][C][solax_meter_gateway:097]:     Accuracy Decimals: 0
[17:53:58][C][solax_meter_gateway:098]:   Operation name 'solax-meter-gateway-esp32 operation mode'
[17:53:58][C][solax_meter_gateway:098]:     Icon: 'mdi:heart-pulse'
[17:53:58][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[17:53:58][C][solax_meter_gateway.switch:068]: SolaxMeterGateway Switch 'solax-meter-gateway-esp32 emergency power off'
[17:53:58][C][solax_meter_gateway.switch:070]:   Icon: 'mdi:power'
[17:53:58][C][solax_meter_gateway.switch:091]:   Restore Mode: restore defaults to OFF
[17:53:58][C][solax_meter_gateway.switch:052]:   Restore Mode: Restore (Defaults to OFF)
[17:53:58][C][mdns:108]: mDNS:
[17:53:58][C][mdns:109]:   Hostname: solax-meter-gateway-esp32
[17:53:58][C][ota:093]: Over-The-Air Updates:
[17:53:58][C][ota:094]:   Address: solax-meter-gateway-esp32.local:3232
[17:53:58][C][ota:097]:   Using Password.
[17:53:58][D][uart_debug:114]: >>> 01:03:02:00:00:B8:44
[17:53:58][C][api:138]: API Server:
[17:53:58][C][api:139]:   Address: solax-meter-gateway-esp32.local:6053
[17:53:58][C][api:141]:   Using noise encryption: YES
[17:53:58][C][mqtt:065]: MQTT:
[17:53:58][C][mqtt:067]:   Server Address: 192.168.168.201:2884 (192.168.168.201)
[17:53:58][C][mqtt:068]:   Username: 'sonoff'[redacted]
[17:53:58][C][mqtt:069]:   Client ID: 'solax-meter-gateway-esp32-c44f336a5301'[redacted]
[17:53:58][C][mqtt:071]:   Discovery prefix: 'homeassistant'
[... 60 lines cut ...]
[17:54:04][D][uart_debug:114]: >>> 01:03:02:00:00:B8:44
[17:54:04][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[17:54:04][D][uart_debug:114]: >>> 01:03:02:00:00:B8:44
[17:54:04][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[17:54:04][D][uart_debug:114]: >>> 01:03:02:00:00:B8:44
[17:54:04][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[17:54:04][D][sensor:094]: 'solax-meter-gateway-esp32 instantaneous power consumption': Sending state 487.17999 W with 2 decimals of accuracy
[17:54:04][D][uart_debug:114]: >>> 01:03:02:00:00:B8:44
[17:54:04][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[17:54:05][D][uart_debug:114]: >>> 01:03:02:00:00:B8:44
[17:54:05][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[17:54:05][D][uart_debug:114]: >>> 01:03:02:00:00:B8:44
[17:54:05][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[17:54:05][D][uart_debug:114]: >>> 01:03:02:00:00:B8:44
[17:54:05][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[17:54:05][D][uart_debug:114]: >>> 01:03:02:00:00:B8:44
[17:54:06][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[17:54:06][D][sensor:094]: 'solax-meter-gateway-esp32 instantaneous power consumption': Sending state 493.38000 W with 2 decimals of accuracy
[17:54:06][D][uart_debug:114]: >>> 01:03:02:00:00:B8:44
[17:54:06][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[17:54:06][D][uart_debug:114]: >>> 01:03:02:00:00:B8:44
[17:54:06][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[17:54:06][D][uart_debug:114]: >>> 01:03:02:00:00:B8:44
[17:54:06][D][uart_debug:114]: <<< 01:03:00:0B:00:01:F5:C8
[17:54:06][D][uart_debug:114]: >>> 01:03:02:00:00:B8:44

dirk99, 2023-07-20:
Die Kommunikation läuft jetzt, war ein Draht abgebrochen :-(. Welchen Wert benötigt der Umrichter jetzt ganz genau, um abzuregeln? Ich habe ihm den 30053_Total_system_power aus dem SDM72 übermittelt. Der Wert wird negativ, wenn ich einspeise und Positiv bei Einspeisung durch den Energieversorger. Ich habe diesen Wert auch schon negiert. Der Solar Regelt, aber geht dann zwischendurch auch auf 0 Produktion, obwohl ich noch vom Energieversorger beziehe. Für mich sieht es so aus, als wenn der Solar irgendwie im Wartemodus steht, aber nicht in der Regelung.

syssi, 2023-07-20:
Deine Beschreibung reicht leider nicht aus um dich zu unterstützen. Kannst du versuchen an einem sonnigen Tag eine konstante Last zu erzeugen? Würde die Regelung nicht arbeiten, würde der Wechselrichter auch keinen Beitrag leistet. Sobald die Regelung einen Bedarf erkennt (das Vorzeichen ist hier in der Tat relevant, ein positiver Smartmeter-Wert bedeutet Import von Leistung aus dem Netz) wird genau dieser Wert an den Wechselrichter kommuniziert. Dieser muss nun hochfahren und diesen Wert liefern. Die aktuelle gelieferte Leistung sollte im Display ablesbar sein. Ein ESPHome-Logauszug würde helfen um zu beurteilen, was die Regelung an Informationen erhält und was sie draus macht.

Wichtig ist der Betriebsmodus "Auto" und der Notaus dart mich als eingeschaltet sein.

dirk99, 2023-08-03:
Sorry hat etwas länger gedauert. Welcher Wert wird genau vom Stromzähler für die Begrenzung benötigt. Der Wert für den Export, Import oder der tatsächliche Verbrauch (Import + Solar Erzeugung). Er regelt jetzt inzwischen, aber ich bekomme den 0 Export nicht hin.

syssi, 2023-08-03:
Der Solax-Inverter / das Gateway möchte über die Menge an Strom informiert werden, welcher deine Stromrechnung in die Höhe treiben würde. In deinen Worten der tatsächliche Verbrauch.

dirk99, 2023-08-03:
Also Import plus Solar ErzeugungMit freundlichen GrüßenDirk HowoldMobil ***@***.***

syssi, 2023-08-03:
Genau!