syssi/esphome-solax-x1-mini #61: Feature Request: Override Switch for Export Control

syssi, 2023-12-08:
### Discussed in https://github.com/syssi/esphome-solax-x1-mini/discussions/60

<div type='discussions-op-text'>

<sup>Originally posted by **Sven177** December  8, 2023</sup>
Having an Override Switch for the Meter Gateway, to toggle between either listening to an existing sensor´s value to determine the current power demand, or using a fixed value (like 500W) would be nice.

This would allow to "disable" the export control function on the Meter Gateway side whenever needed, while leaving the actual export control function of the Inverter (and according settings) still active and untouched.

(Hope this is the right way for a Feature Request - first time for me) 😉</div>

syssi, 2023-12-29:
@Sven177 The feature is ready. Do you like to give it a try?

Sven177, 2023-12-29:
Awesome, thanks!
I´ll give it a try...   👍

Sven177, 2023-12-29:
Was trying to use that manual switches in the ESP32 multiple UART code, but it throws me that one when adding that underneath the emergency off switches: 

_[manual_mode] is an invalid option for [switch.solax_meter_gateway]. Please check the indentation._

Am I missing something?

syssi, 2023-12-29:
The new feature isn't merged into the `main` branch yet. Did you change the external component section to the feature branch?

Sven177, 2023-12-29:
Ah, no. That is still @main.
So change that as well?

Sven177, 2023-12-29:
**Awesome!**
Both switches and both Numeric Input Fields show up as entities already.
Will test tomorrow, when Inverter(s) are up and running, and will report (but looking fine so far)   :-)

Sven177, 2023-12-30:
Seems working. 👍 
Had one Meter Fault as operation mode from the gateway, after switching back from manual to auto-mode, but couldn´t reproduce it yet when switching modes...

syssi, 2023-12-30:
The short meter fault while turning off the manual mode is fine (and harmless) because the implementation does need a cycle to pick up the measurement of your smart meter and make it available as valid power demand.

closed 2023-12-30 by syssi

syssi, 2023-12-30:
I've prepared a new release. You should switch back to the `@main` branch again.

PR #64 (2023-12-30): Introduce manual mode to set a static power demand
Closes: #61