syssi/esphome-solax-x1-mini #62: Still problems with sporadic meter faults

Sven177, 2023-12-12:
Hi Sebastian,

the export control is basically working, and I am using 2x D1 Mini Boards for 2 separate inverters.
But sometimes, I get the meter fault error thrown by the inverter and I haven´t yet figured out why.
I found, that the source sensor for power consumption (Shelly 3EM) sometimes takes 10-15 seconds to update values regaridng power consumption. To get around this, I added an heartbeat filter to the code, to send the source value to the power demand sensor every (tried) 3 or 5 seconds, even if the power consumtion sensor hasn´t been updated in between.

```substitutions:
  name: solax-meter-gateway
  device_description: "Control the export of a Solax X1 mini via RS485"
  external_components_source: github://syssi/esphome-solax-x1-mini@main

external_components:
  - source: ${external_components_source}
    refresh: 0s

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: GPIO12
    rx_pin: GPIO14

solax_meter_modbus:
  - id: modbus0
    uart_id: uart_0

solax_meter_gateway:
  solax_meter_modbus_id: modbus0
  address: 0x01
  # The state of this sensor (instantaneous power in watt) is used as source
  power_id: powermeter0
  power_sensor_inactivity_timeout: 25s
  update_interval: 1s

sensor:
  - platform: homeassistant
    id: powermeter0
    entity_id: sensor.power_consumption
    internal: true
    force_update: true
    filters:
      heartbeat: 5s

  - platform: solax_meter_gateway
    power_demand:
      name: "${name} power demand"

text_sensor:
  - platform: solax_meter_gateway
    operation_mode:
      name: "${name} operation mode"
´´´

But despite that change, sporadically the meter fault still shows up.
Any ideas what would cause this? Could it be related to using 2 Boards at 2 Inverters sharing the same code?

Out in the field currently (work related), will be back on Thursday to provide according logs or increase the Log Level

Sven177, 2023-12-14:
Hi Sebastian,

after checking the Log, I doubt that the heartbeat would do anything different, since the last value of that source sensor is handed over very frequently to the inverter.

```[09:48:38][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:38][D][sensor:093]: 'powermeter0': Sending state 238.89999  with 1 decimals of accuracy
[09:48:39][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 238.89999 W with 0 decimals of accuracy
[09:48:39][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:39][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 238.89999 W with 0 decimals of accuracy
[09:48:39][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:39][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 238.89999 W with 0 decimals of accuracy
[09:48:39][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:39][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 238.89999 W with 0 decimals of accuracy
[09:48:39][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:40][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 238.89999 W with 0 decimals of accuracy
[09:48:40][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:40][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 238.89999 W with 0 decimals of accuracy
[09:48:40][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:40][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 238.89999 W with 0 decimals of accuracy
[09:48:40][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:40][D][sensor:093]: 'powermeter0': Sending state 238.89999  with 1 decimals of accuracy
[09:48:40][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 238.89999 W with 0 decimals of accuracy
[09:48:40][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:41][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 238.89999 W with 0 decimals of accuracy
[09:48:41][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:41][D][homeassistant.sensor:024]: 'sensor.power_consumption': Got state 249.08
[09:48:41][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 238.89999 W with 0 decimals of accuracy
[09:48:41][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:41][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 238.89999 W with 0 decimals of accuracy
[09:48:41][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:42][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 238.89999 W with 0 decimals of accuracy
[09:48:42][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:42][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 238.89999 W with 0 decimals of accuracy
[09:48:42][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:42][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 238.89999 W with 0 decimals of accuracy
[09:48:42][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:42][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 238.89999 W with 0 decimals of accuracy
[09:48:42][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:42][D][sensor:093]: 'powermeter0': Sending state 249.08000  with 1 decimals of accuracy
[09:48:43][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 249.08000 W with 0 decimals of accuracy
[09:48:43][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:43][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 249.08000 W with 0 decimals of accuracy
[09:48:43][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:43][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 249.08000 W with 0 decimals of accuracy
[09:48:43][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:43][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 249.08000 W with 0 decimals of accuracy
[09:48:43][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:44][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 249.08000 W with 0 decimals of accuracy
[09:48:44][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:44][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 249.08000 W with 0 decimals of accuracy
[09:48:44][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:44][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 249.08000 W with 0 decimals of accuracy
[09:48:44][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[09:48:44][D][sensor:093]: 'powermeter0': Sending state 249.08000  with 1 decimals of accuracy
[09:48:44][D][sensor:093]: 'solax-meter-gateway power demand': Sending state 249.08000 W with 0 decimals of accuracy
[09:48:44][D][text_sensor:064]: 'solax-meter-gateway operation mode': Sending state 'On'
[... cut ...]
The manual says wrong wiring / wrong baud rate for troubleshooting, but I would doubt those 2 reasons for this case - since it works most of the time but just sporadically gets to that meter error state.

PS: also running kind of the same config at my Dad´s Inverters. Also 2 pcs. D1 Mini Boards, completely independently wired and connected. Where also sporadically the meter fault shows up and I don´t know exactly what´s causing it.
Even though the inverter is recovering from that state usually by itself, it would be of course better to not fall into that meter fault state at all... 😅😁

syssi, 2023-12-14:
@Sven177 Do you use discord and can we have a quick chat here? My handle is `syssi.`

Sven177, 2023-12-14:
I have a discord account - but how to find you there? 😅

syssi, 2023-12-14:
Direct messages -> Add friend -> `syssi.`

Sven177, 2023-12-14:
seems I found a syssi - but it seems it is the wrong one.
Oh, the "." needs to be there as well?

syssi, 2023-12-14:
Long story short: Using hardware UARTs (GPIO1/GPIO3 on an ESP8266 or any GPIO on an ESP32) should resolve the issue.

Sven177, 2023-12-14:
OK, tried using GPIO1 and GPIO3 on ESP8266, but getting meter fault instantly on that config... 🤔

syssi, 2023-12-14:
Did you swap RX/TX accidently? Did you update the `uart` section of your YAML configuration? Does the TX led of the RS485 converter board blink periodically?

Sven177, 2023-12-14:
Hi,

yes - here from the Log. Is the internal UART missing some adjustment for the inverter?

```
[11:51:01][C][logger:416]: Logger:
[11:51:01][C][logger:417]:   Level: DEBUG
[11:51:01][C][logger:418]:   Log Baud Rate: 0
[11:51:01][C][logger:420]:   Hardware UART: UART0
[11:51:01][C][uart.arduino_esp8266:102]: UART Bus:
[11:51:01][C][uart.arduino_esp8266:103]:   TX Pin: GPIO1
[11:51:01][C][uart.arduino_esp8266:104]:   RX Pin: GPIO3
[11:51:01][C][uart.arduino_esp8266:106]:   RX Buffer Size: 256
[11:51:01][C][uart.arduino_esp8266:108]:   Baud Rate: 9600 baud
[11:51:01][C][uart.arduino_esp8266:109]:   Data Bits: 8
[11:51:01][C][uart.arduino_esp8266:110]:   Parity: NONE
[11:51:01][C][uart.arduino_esp8266:111]:   Stop bits: 1
[11:51:01][C][uart.arduino_esp8266:113]:   Using hardware serial interface.
[11:51:01][C][solax_meter_modbus:096]: SolaxMeterModbus:
```

Sven177, 2023-12-18:
Hi,

seems like for whatever reason my bord doesn´t like that Pin 1 and 3 at all, so I was using the SWAP Config, to assign Pins 13 and 15 instead.

```
substitutions:
  name: solax-meter-gateway
  device_description: "Control the export of a Solax X1 mini via RS485"
  external_components_source: github://syssi/esphome-solax-x1-mini@main

external_components:
  - source: ${external_components_source}
    refresh: 0s

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: GPIO15
    rx_pin: GPIO13
    debug:
      direction: BOTH    

solax_meter_modbus:
  - id: modbus0
    uart_id: uart_0

solax_meter_gateway:
  solax_meter_modbus_id: modbus0
  address: 0x01
  # The state of this sensor (instantaneous power in watt) is used as source
  power_id: powermeter0
  power_sensor_inactivity_timeout: 20s
  update_interval: 1s
```

It was working on that config so far.
Nevertheless, I decided to switch to an ESP32 Board, for controlling 2 Inverters with one of those at the same time, instead of using two separate 8266 boards for that purpose... 😉

closed 2023-12-18 by Sven177