syssi/esphome-solax-x1-mini #83: Meter Gateway

dirk99, 2025-07-27:
Hallo, ich habe seit einiger Zeit das "esphome-modbus-solax-x1" in Benutzung. Hast Du das Projekt eingestellt, oder kann diese Project auch die Funktion? Kannst Du mir ggf. den letzten Code vom esphome-modbus-solax-x1 bereitstellen, mir ist der ESP abgeraucht...? Danke im Voraus

syssi, 2025-07-27:
Diese Projekt gibt es weiterhin. Hast du die YAML-Konfiguration nicht mehr, welche du damals auf den ESP geschrieben hast? Sie wird sehr individuell auf dein Setup zugeschnitten gewesen sein.

dirk99, 2025-07-27:
Die Url aus meiner yaml ist nicht erreichbar und ich kann das Projekt bei Dir im Git auch nicht mehr finden. Kannst Du die url hier mal Posten. Danke